RIOT-OS/RIOT #7719: saul_reg: const pointer for device?

vincent-d, 2017-10-11:
#7586 changed the `dev` field of `saul_reg_t` fron `void *` to `const void *`.

I don't see why it should be const, some driver might need to update the device descriptor while using it.
It actually break some of my code where the driver needs to update the descriptor.

So what is the rationale for this change?

cladmi, 2017-10-11:
The rational was allowing the dev structures to be stored in rom, but should not constraint the device.

From my knowledge, it does not require the structure to be const. It only prevents uses of the saul_reg device to modify it through this pointer https://stackoverflow.com/a/31109590.
Please correct me if I am wrong here.
In the same way parameters can be defined as `const` pointers only as a contract that the function will not modify the pointed data. As saul read/write functions cannot modify the device there is no need to have a writeable device.

So you should still be able to modify your device outside of saul.

vincent-d, 2017-10-11:
Understood, but this code does not work anymore:
```C
saul_reg_t *adc = saul_reg_find_type(SAUL_SENSE_ANALOG);
if (adc) {
    adcxx1c_enable_alert(adc->dev, adc_alert_cb, NULL);
}
```
I am going to need to cast explicitly `adc->dev` then.

haukepetersen, 2017-10-12:
Just had an offline discussion with @cladmi: making the device descriptor pointer non-const again seems to be the way for us -> see #7723

closed 2017-10-12 by vincent-d

cladmi, 2017-10-12:
I did not know that a `const dev_t *dev` pointer could be casted to `(void *)` without discarding `const`. That is why I chose this solution with side effects.

PR #7723 (2017-10-12): saul_reg: make ptr to device descriptor non const
fixes #7719

Rationale: all devices are identified by their device descriptor, which points to the run-time data of devices. So typically, the device descriptor points into RAM (using a non-const pointer). Now for some devices, we have the case that they don't need/have any dynamic run-time data (e.g. the SAUL gpio driver). For these devices we can optimize their memory usage by keeping their device descriptor in ROM (e.g. by using their static configuration data as device descriptor). This would call for their device descriptor being a const p [... cut]