RIOT-OS/RIOT #7721: arpa/inet.h: No such file or directory

kbumsik, 2017-10-12:
Hi,
When I try compiling `tests/driver_enc28j60` with the following command, 

> BOARD=stm32f4discovery USEMODULE+=gnrc_uhcpc make all

I got an error:

```
RIOT/sys/net/application_layer/uhcp/uhcp.c:11:23: fatal error: arpa/inet.h: No such file or directory
 #include <arpa/inet.h>
                       ^
compilation terminated.
```
Is it unexpected error, or did I miss something? I would appreciate if someone let me know.
Thank you.

miri64, 2017-10-12:
Can you try with `USEMODULE += posix`?

miri64, 2017-10-12:
For me it is working with it, but there seems to be another dependency problem.

miri64, 2017-10-12:
See https://github.com/RIOT-OS/RIOT/pull/7722 to fix that definitively.

kbumsik, 2017-10-12:
It compiles correctly with your fix in #7722 .
Thank you for a quick response :+1:

kbumsik, 2017-11-06:
I will close this issue since #7722  is merged.

closed 2017-11-06 by kbumsik

miri64, 2017-11-08:
Sorry we needed to revert that (see #7967), since that merge was against the GNRC merge-embargo. But it is fixed in #7925.