RIOT-OS/RIOT #7734: stm32: approach for common peripheral definitons

nmeum, 2017-10-13:
While porting RIOT to the bluepill port #7727 we noticed that there are many boards supported by RIOT which use the stm32f103 MCU. Since many definitions we added and tested are similar among the boards which use this MCU we wonder if it would make sense to implement a common `periph_conf.h` header file for those boards.

For instance if you compare the `periph_conf.h` header file of the nucleo-f103 board with the one from the maple-mini they look eye to eye identical. The problem with this code duplication is the following: When porting RIOT to the bluepilll board we found that some peripherals were not configured properly, for instance the general purpose timer four wasn't configured. Instead of backporting the changes made by us for the bluepill board we would suggest adding some kind of common `periph_conf.h` for these boards.

Any suggestion on how to do this? We noticed the `-common` board folders in `boards/`, however, they don't seem to fit our purpose sense these are common folders for boards made by one manufacturer.

temmihoo, 2017-11-14:
In the MCU world most "board" peripherals are actually integrated within the MCU itself by the silicon vendor.

Often the only board-specific part of configuration are the board pin names. Some boards do have external-to-MCU hardware features such as early ELL-i boards having ENC28J60 and quite many boards having a number of LEDs and switches not to mention simple sensors and the like. Supporting these easily without necessitating MCU integrated peripheral support repetition would be ideal.

I think it would make sense to have a commons system that would allow (not exhaustive list):
- things common to a specific MCU model such as STM32F334R8
- things common to a MCU model such as STM32F334*
- things common to a MCU family such as STM32F3* or even STM32F*
- things common to a MCU core such as Cortex M3
- things common to a board family such as ST Nucleo64-*

Personally I tend to think of the board specific pin names as mostly useless abstraction and am always thinking from point of view of the MCU model specific reference manual. This is so because my aim is to help in designing products out of the chips instead of developing endlessly on the push pin wire level. Also, when reading of the hardware peripherals I am reading the MCU model specific reference manual instead of the board vendor slideware.

Achieving this type of orthogonality might be technically difficult using Makefiles and #include directives so some sort of preprocessing might be needed.

temmihoo, 2017-11-14:
#7661 is related

haukepetersen, 2017-11-15:
@temmihoo the commons system you mention is more or less what we do today. There are indeed examples where this is done sloppy and sometimes its hard to tell, but using the most generic implementations possible and splitting concerns between CPU internals and board configuration options is already done. (e.g. periph_conf.h vs periph_cpu.h vs. periph_cpu_common.h etc).

@nmeum I fear it is not quite as trivial as it seems. In general yes: we should of course use shared code/configuration as much as possible and try to remove any code duplication. But in the case of boards, that logically have nothing to do with each other than the CPU used (bluepill vs nulceo), it gets interesting. Of course we could put a common periph_conf.h somewhere for these two (or maybe others as well). But soon we start to have a large number of partly overlapping shared files, and it gets very very hard be aware of side-effects when changing anything in that mess.

For boards that are related to each other (e.g. the nucleos, the remotes) this is a different picture, as there is a clear structure and hierarchy to their shared files. Here there is indeed a lot room for improvement (just naming #7975 as example).

So in the end it seems to me like we have to find a way in the middle between keeping code duplication to a minimum and keeping a clear separation of concerns between different board configurations.

temmihoo, 2017-11-16:
@haukepetersen is there a guideline document on how this structure should be applied? As opposed to reading the source files, not all of which qualify as examples of well applied principle.

Also, I personally think the peripherals in MCU should be in MCU-related files and specifically NOT in any of the board-related files. Putting MCU integrated stuff in board-related files leads to either repetition of peripheral code to all boards using that MCU or missing features from boards.

About the Nucleo families as examples, most Nucleo boards have just the ST-Link and the MCU and nothing else, so their board-level files should be relatively empty and at most have a logical pin name mapping for board-name pins. This naming already gets somewhat complicated as Nucleo boards have both Morpho pin names and Arduino pin names. Still the MCU pin names exist and should also be supported.

haukepetersen, 2017-11-16:
> is there a guideline document on how this structure should be applied?

Not really a good one (yet). What we do have is a rough (but incomplete and very high level) [porting guide](https://github.com/RIOT-OS/RIOT/wiki/Porting-Guide) and we are currently in the process of writing a paper giving some more insights on the high-level concepts involved.

maribu, 2018-08-15:
@temmihoo, @nmeum: Could the (already merged) PR be a starting point for further code de-duplication? https://github.com/RIOT-OS/RIOT/pull/9126

closed 2019-09-10 by stale[bot]