RIOT-OS/RIOT #7745: Can't write MAC address of ENC28J60

maitysubhasis, 2017-10-18:
I can't update the MAC address of enc28j60.   
I am using the following code.

```c++
int main(void)
{
    /* we need a message queue for the thread running the shell in order to
     * receive potentially fast incoming networking packets */
    msg_init_queue(_main_msg_queue, MAIN_QUEUE_SIZE);
    puts("Test application for the enc28j60 driver\n");
    puts("This test just pulls in parts of the GNRC network stack, use the\n"
         "provided shell commands (i.e. ifconfig, ping6) to interact with\n"
         "your enc28j60 device.\n");
         
    uint8_t mac_to_set[6] = {0x00, 0x04, 0xA3, 0x76, 0x09, 0xAA};
    uint8_t mac[6];
    enc28j60_t enc28j60_dev = dev[1]; // get the enc28j60 device
    mac_set(&enc28j60_dev, mac_to_set);
    mac_get(&enc28j60_dev, mac);

    for(int i=0;i<6;i++){
        printf("MAC Byte %i: %u : %u \n", i, mac[i], mac_to_set[i]);
    }
    
    //cmd_w_phy(dev, REG_PHY_PHLCON, PHLCON_LACFG(0xa)|PHLCON_LBCFG(0xa));

    /* start shell */
    puts("Starting the shell now...");
    char line_buf[SHELL_DEFAULT_BUFSIZE];

    shell_run(NULL, line_buf, SHELL_DEFAULT_BUFSIZE);
    /* should be never reached */
    return 0;
}
```

haukepetersen, 2017-11-30:
hi, sorry for the late reply! This could have many reasons. But looking at your snip, I am first wondering why you include the drivers c-files? I guess you do this because your compiler complained that it could not find the functions otherwise?

So please try to remove the c-file-includes and simply add the `enc28j60` module to your application Makefile -> `USEMODULE += enc28j60`.

If this does not help, enable debugging for the `enc28j60` driver (set ENABLE_DEBUG to 1 in the drivers C file), and check the output when you run your application on the node.

PeterKietzmann, 2017-12-18:
@maitysubhasis what's the latest? I just used that device with a nucleo board which worked as expected. As @haukepetersen indicated you need to include that module to your Makefile. This will automatically initialize the device which is (obviously) needed. Furthermore, how about using the [`netapi`](https://riot-os.org/api/group__net__gnrc__netapi.html#gabc41a5d60228a68daa7ef14b0bbb6ff4) interface to set device parameters?

haukepetersen, 2018-01-29:
@maitysubhasis I hope your issue has been resolved.

Since there is no activity  in this issue, I am closing this now. Please re-open if the issue still exists.

maitysubhasis, 2018-04-24:
I still can't make the test work. I think there is some problem in function transmit part of `spi_transfer_bytes` function. And another question why `dev(bus)->DR;` needs to be flushed when busy. Wouldn't it flush received data also?

haukepetersen, 2018-05-24:
sorry for only noticing your reply now, as the issue still exists I reopen it. Don't have the hardware at hand right now, but will give this a try once I am back at work the week after the next.

haukepetersen, 2018-05-24:
oh, but still one more important question: @maitysubhasis which board are you using?

haukepetersen, 2018-05-24:
hmm, did look at my backlog in the wrong order. So looking at #9008 I am confused now: does this issue still exist or not?

closed 2019-01-20 by maitysubhasis