RIOT-OS/RIOT #7753: pic32-wifire: race-condition when linking in concurrent build

miri64, 2017-10-18:
When trying to build an application utilizing `xtimer` or `pm` multithreaded (using the `-j <n>` parameter for `make`; our buildtests do that) I sometimes (sadly not always, but I was able to reproduce it multiple times) get a linking error (I build `gnrc_networking` in this example):

```
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/shell_commands.a(sc_random.o): In function `_random_init':
sc_random.c:(.text._random_init+0xc): undefined reference to `timer_read'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/shell_commands.a(sc_icmpv6_echo.o): In function `.L61':
sc_icmpv6_echo.c:(.text._icmpv6_ping+0x1f8): undefined reference to `timer_read'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/shell_commands.a(sc_icmpv6_echo.o): In function `.L63':
sc_icmpv6_echo.c:(.text._icmpv6_ping+0x250): undefined reference to `timer_read'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/xtimer.a(xtimer_core.o): In function `_xtimer_lltimer_now':
../include/xtimer/implementation.h:(.text._xtimer_lltimer_now+0x0): undefined reference to `timer_read'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/xtimer.a(xtimer_core.o): In function `_lltimer_set':
../include/xtimer/implementation.h:(.text._lltimer_set+0xa): undefined reference to `timer_set_absolute'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/xtimer.a(xtimer_core.o): In function `xtimer_init':
../include/xtimer/implementation.h:(.text.xtimer_init+0x16): undefined reference to `timer_init'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/xtimer.a(xtimer.o): In function `_xtimer_spin':
xtimer.c:(.text._xtimer_spin+0x6): undefined reference to `timer_read'
xtimer.c:(.text._xtimer_spin+0xe): undefined reference to `timer_read'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/core.a(kernel_init.o): In function `.L4':
kernel_init.c:(.text.idle_thread+0x4): undefined reference to `pm_set_lowest'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/bin/pic32-wifire/gnrc_rpl.a(gnrc_rpl_dodag.o): In function `.L45':
gnrc_rpl_dodag.c:(.text.gnrc_rpl_parent_remove+0x3e): undefined reference to `timer_read'
collect2: error: ld returned 1 exit status
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/gnrc_networking/../../Makefile.include:306: recipe for target 'link' failed
make: *** [link] Error 1
```

kYc0o, 2017-11-22:
Was this maybe solved by #6892 as it solved #6588?

miri64, 2017-11-22:
Maybe. Will run the buildtests tonight.

miri64, 2017-11-22:
Can't test tonight... have to run longterm tests for #7925. But will do ASAP.

neiljay, 2017-11-24:
Its more likely to be fixed by this:  #8032

cladmi, 2017-11-27:
If you can reproduce it, I am interested in the result of

```
ar t bin/pic32-wifire/periph.a
```

I should only have a subset of the object files

```
pm.o
timer.o
uart.o
cpuid.o
gpio.o
hwrng.o
```

If its the case, then yes it should be solved by #8032

Manually deleting `periph.a` also brings me a broken build output because different modules use the same name.

cladmi, 2017-11-30:
@miri64 do you have news for this one ?

miri64, 2017-11-30:
Gnarf... sorry I totally forgot about this and it doesn't make much sense to run it now, since I'm on vacation as of tomorrow until Sunday. So I'll take a note to test it Monday.

miri64, 2017-12-05:
Done the tests now: There is still some race-condition for the `pic32-wifire`.

miri64, 2017-12-05:
After some repeated forced builds on `examples/default` here is the output:

```
Building application "default" for "pic32-wifire" with MCU "mips_pic32mz".

"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/boards/pic32-wifire
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/core
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips_pic32mz
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips32r2_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips_pic32_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips32r2_common/newlib_syscalls_mips_uhi
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips32r2_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/periph_common
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/cpu/mips_pic32_common/periph
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/drivers/saul
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/auto_init/saul
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/fmt
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/isrpipe
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/phydat
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/ps
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/saul_reg
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/shell
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/shell/commands
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/tsrb
"make" -C /home/mlenders/Repositories/RIOT-OS/RIOT/sys/uart_stdio
   text    data     bss     dec     hex filename
 114380    2792    6028  123200   1e140 /home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/bin/pic32-wifire/default.elf
objcopy: --change-section-lma .gcc_except_table+0xffffffff80000000 never used
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default'
make: Entering directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default'
/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default/../../Makefile.include:592: /home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/mcuboot.mk: No such file or directory
make: *** No rule to make target '/home/mlenders/Repositories/RIOT-OS/RIOT/makefiles/mcuboot.mk'.  Stop.
make: Leaving directory '/home/mlenders/Repositories/RIOT-OS/RIOT/examples/default'
```

cladmi, 2017-12-05:
So it correctly compiled youhou (the `--change-section-lma .gcc_except_table+0xffffffff80000000 never used` was already there).

But why is it re-entering `examples/default` and not finding `mcuboot.mk` ?

miri64, 2017-12-05:
> But why is it re-entering examples/default and not finding mcuboot.mk ?

Maybe I chose the wrong cut-off ^^ my command was basically:

```sh
while true; do make -j 8 -C examples/default clean all || break; done
```

So I guess the previous lines were just from a previous build ^^"

miri64, 2017-12-05:
And: maybe this error doesn't even has anything to do with the race-condition :-/. But the race-condition is still there, when building with `dist/tools/compile_test/compile_test.py`.

francois-berder, 2020-01-31:
I tried hard to reproduce this issue but I never encountered this bug. I recently left my machine overnight running this command:

```sh
while true; do BOARD=pic32-wifire make -j 8 -C examples/default clean all || break; done
```

and yet I was not able to trigger this bug. So I really wonder if this bug actually exists now given that https://github.com/RIOT-OS/RIOT/pull/8032 was merged.

I have not tried with `dist/tools/compile_test/compile_test.py` though.

miri64, 2020-01-31:
@fjmolinas you most likely made a few builds for the current release. Can you confirm this? I think we can close this if so. Both the mips platform and the build system received some improvements over the last few months.

miri64, 2020-07-01:
Tried to reproduce on my machine again, but https://github.com/RIOT-OS/RIOT/issues/14410 got in my way. Trying with docker now:

```sh
while true
do
    BOARD=pic32-wifire BUILD_IN_DOCKER=1 DOCKER_MAKE_ARGS=-j16 \
        make -C examples/default clean all || break;
done
```

miri64, 2020-07-01:
I ran this loop now for >2h and did not run into that race condition. I think we can safely assume this is not an issue anymore.

closed 2020-07-01 by miri64