RIOT-OS/RIOT #7757: Linker does not link the auto_init files

florian-popup, 2017-10-18:
Hello,

I m trying to use the gnrc_networking example with a custom board using an samd21 and mrf24j40.

but when I try to implement the mrf24j40 drivers, I get

> auto_init.c:180: undefined reference to `auto_init_mrf24j40'

So apparently the compiler doesn't take into account the sys/auto_init/netif dir.
I tried to include this dir directly in my board makefile without success

my Makefile for the board :
`MODULE = board

USEMODULE += mrf24j40
USEMODULE += gnrc_netdev_default
USEMODULE += auto_init_gnrc_netif

include $(RIOTBASE)/Makefile.base`

I also added to the board.h (does it belong here ?)

`#define MODULE_MRF24J40`

What am I missing ?
Thanks for your help

aabadie, 2017-10-18:
In the gnrc_networking example Makefile, you also  have to add your board to the list of netif boards. There's a variable for that.

florian-popup, 2017-10-18:
The board selection was already made but apparently I was doing it wrong because I modified the board makefile and not the application makefile
I mostly followed the interesting step there https://github.com/RIOT-OS/RIOT/wiki/Tutorial:-How-to-port-a-radio-module-driver-to-RIOT-OS

In short I just added a conditional inclusion of radio module in the board makefile.dep
And put USEMODULE += mrf24j40 in the application makefile.
AND REMOVED THE  `#define MODULE_MRF24J40` line that was giving me the error

Now I get no more response from the UART when I include the radio module but only it's own debug. I guess it's more likely to be a problem with my hardware configuration

jia200x, 2017-10-20:
Hi @florian-popup 

In case you are adding your custom board and the radio is embedded, you can add the mrf24j40 module in board's Makefile.dep ([E.g for Samr21-xpro](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/Makefile.dep). Otherwise, just in the application Makefile should be fine.

This will automatically define the MODULE_MRF24J40 directive.

The sys/auto_init/netif folder will be added automatically if USEMODULE += auto_init_gnrc_netif is included in Makefiles.

Can you tell us more about the hardware configuration? Are you setting the correct pins for the radio?

florian-popup, 2017-10-23:
@jia200x 

Thanks for your help, that's what I figured out.
I made a conditional inclusion of the mrf24j40 module in my board makefile definition if the auto_init_gnrc_netif and it compiles.

I get the UART working 
`static const uart_conf_t uart_config[] = {
    {    /* Virtual COM Port */
        .dev      = &SERCOM0->USART,
        .rx_pin   = GPIO_PIN(PA,9),
        .tx_pin   = GPIO_PIN(PA,8),
        .mux      = GPIO_MUX_C,
        .rx_pad   = UART_PAD_RX_1,
        .tx_pad   = UART_PAD_TX_0,
        .flags    = UART_FLAG_NONE,
        .gclk_src = GCLK_CLKCTRL_GEN_GCLK0
    }
};
`

and I configured the SPI
`static const spi_conf_t spi_config[] = {
    {   /* EXT2 */
        .dev      = &SERCOM3->SPI,
        .miso_pin = GPIO_PIN(PA, 25),
        .mosi_pin = GPIO_PIN(PA, 22),
        .clk_pin  = GPIO_PIN(PA, 23),
        .miso_mux = GPIO_MUX_C,
        .mosi_mux = GPIO_MUX_C,
        .clk_mux  = GPIO_MUX_C,
        .miso_pad = SPI_PAD_MISO_3,
        .mosi_pad = SPI_PAD_MOSI_0_SCK_1
    }
};`

I made sure the SPI pins doesn't conflict with any other pins declared in the periph_conf.h but the SPI communication fail to initialize. 
Could it be because I defined the CS pin of the mrf24j40 board as the hardware CS pin of the matching SPI communication pins of my microcontroller. (Instead of a random GPIO port ?) 

`/**
 * @name   mrf24j40 configuration
 * @{
 */

#define MRF24J40_PARAM_SPI          (SPI_DEV(0))
#define MRF24J40_PARAM_SPI_CLK      (SPI_CLK_5MHZ)
#define MRF24J40_PARAM_CS           (GPIO_PIN(PA, 24))
#define MRF24J40_PARAM_INT          (GPIO_PIN(PA, 28))
#define MRF24J40_PARAM_RESET        (GPIO_PIN(PA, 19))

#define MRF24J40_PARAMS_BOARD     { .spi = MRF24J40_PARAM_SPI, \
                                      .spi_clk = MRF24J40_PARAM_SPI_CLK, \
                                      .cs_pin = MRF24J40_PARAM_CS, \
                                      .int_pin = MRF24J40_PARAM_INT, \
                                      .reset_pin = MRF24J40_PARAM_RESET }

#define MRF24J40_PARAM_WAKE            (GPIO_PIN(PA, 27))`

I tried to make this work by both initializing manually or not initializing the GPIO MRF24J40_PARAM_INT/CS/RESET in the init board function. Both failed to communicate.
Should initialise the GPIO (in or out) myself for the pin connected to the mrf24j40 ?

Thanks

closed 2017-10-25 by florian-popup