RIOT-OS/RIOT #7767: No default fall back when TERMPROG is not set

gebart, 2017-10-19:
#7758 broke my config because my environment has TERMINAL=urxvt set and there is no else clause in the ifeq block and there is no fall back if TERMPROG ends up being empty.

Can we change TERMINAL to something else? Some Unix environments (e.g. i3) expect $TERMINAL to be the preferred graphical terminal.

On a side note: `command -v` in bash without any argument returns 0 as error code, which prevents the error message from appearing. `command -v ''` on the other hand gives 1 as status. In zsh, `command -v` with no argument gives 1 as status code.

miri64, 2017-10-23:
Fixed in #7789.

miri64, 2017-10-23:
> On a side note: command -v in bash without any argument returns 0 as error code, which prevents the error message from appearing. command -v '' on the other hand gives 1 as status. In zsh, command -v with no argument gives 1 as status code.

I don't understand how this is related to the rest at all :-/.

gebart, 2017-10-24:
> I don't understand how this is related to the rest at all :-/.

I meant that the check for TERMPROG existence using `command -v` at https://github.com/RIOT-OS/RIOT/blob/master/Makefile.include#L386 reports that the TERMPROG command is working, even though the variable is empty, so no error message is displayed and the make script tries to execute some weird command line based on only TERMFLAGS.

I would have expected that I would see an error message saying `Terminal program  not found. Aborting.` instead of nothing happening.

PR #7789 (2017-10-24): make: prefix TERMINAL macro
Fixes #7767 by prefixing the generic `TERMINAL` variable (which might be required to be defined in the environment by window managers like `i3`) with `RIOT_`.
commit e3e1abb3 (2017-10-24): make: prefix TERMINAL macro

Fixes #7767
commit 764e4089 (2017-10-24): make: prefix TERMINAL macro

Fixes #7767
closed 2017-10-24 by kaspar030