RIOT-OS/RIOT #7800: cbor: fix module

miri64, 2017-10-24:
Last release tests found some quality defects in the `cbor` module (some of them known, some of them not). The following things need to happen:

- [ ] make sure the implementation is compliant with RFC 7049 (according to #7708 it is not)
- [ ] fix issues found by fuzzing (was #7708)
- [ ] move formatting to `fmt` based implementation (should fix #1891 partly)
- [ ] make sure POSIX-based functions are available on non-native platforms (was #7287 an #7411)
- [ ] remove recursion were possible
- [ ] clean-up code

Alternative approach to "fix" this issue would be to just remove it as suggested in #7708.

gebart, 2017-11-13:
There's an alternative implementation for very constrained devices available at https://github.com/cabo/cn-cbor which may be an option if this module is dropped.

kaspar030, 2017-11-13:
> There's an alternative implementation for very constrained devices available at https://github.com/cabo/cn-cbor which may be an option if this module is dropped.

It packaged nicely: #8025

gebart, 2017-11-13:
@kaspar030 cool!

kaspar030, 2018-04-16:
We should mark this module "deprecated" somehow...

miri64, 2018-04-16:
Use `@deprecated` command in `sys/include/cbor.h` for that.

kaspar030, 2018-04-17:
> Use @deprecated command in sys/include/cbor.h for that.

That doesn't warn when compiling, right?

kaspar030, 2018-04-20:
I've added a note to the release notes.

kYc0o, 2018-08-15:
Now that the module is completely deprecated (for 2 release cycles), can we close this issue?

miri64, 2018-08-15:
It's still not removed... So I'd vote against it, but I wouldn't block a close either, if people think it should be closed.

miri64, 2018-08-15:
Ah, but we can remove it now :-)

kYc0o, 2018-08-15:
> Ah, but we can remove it now :-)

👍 thumbs up to remove it!

miri64, 2018-08-15:
> +1 thumbs up to remove it!

Already asked @nmeum to rebase #8793 ;-)

commit e523e8f5 (2018-09-25): sys/cbor: remove module

Fixes #7707
Fixes #7800
closed 2018-09-25 by miri64