RIOT-OS/RIOT #7801: posix: missing functionality

kaspar030, 2017-10-24:
- [ ] getaddrinfo() (see #841)
- [ ] (p)select() (see #4222)

kaspar030, 2017-10-24:
- [ ]￼ (p)select() (see #4222)

miri64, 2017-10-24:
?

kaspar030, 2017-10-24:
> ?

(there was a test comment which I deleted) I wanted to see if checkboxes in comments are possible and whether they're counted as if they're in the description. That would've allowed us to "vote" priority using github reactions. Unfortunately, it doesn't work that way.

miri64, 2019-08-10:
> getaddrinfo() (see #841)

Can be implemented using `sock_dns`.

> (p)select() (see #4222)

Requires asynchronous sock (see #11723)

miri64, 2020-02-11:
`select()` is implemented in #12975.

HendrikVE, 2021-02-09:
+1 for an implementation of `getaddrinfo` for #15969

miri64, 2021-02-09:
> > getaddrinfo() (see #841)
> 
> Can be implemented using `sock_dns`.

This still should hold up true