RIOT-OS/RIOT #7828: Spi communication with mrf24j40 fail but board can run test_spi sucessfully

florian-popup, 2017-10-25:
Hello,

I have a custom board with an atsamd21e18a and a mrf24j40ma as transceiver.
I can't get to make the communication working between those devices.

Here is the hardware configuration of my board
`
(samd21e18a) to (mrf24j40)
PA25 (physical pin 24) to SDO
PA24 (physical pin 23) to CS
PA23 (physical pin 22) to SCK
PA22 (physical pin 21) to SDI
PA19 (physical pin 20) to Reset
PA28 (physical pin 27) to INT
PA27 (physical pin 25) to Wake
`
Correspondance between the PAx and physical pin x have been triple checked.

So far, my board.h configuration concerning the mrf24j40

`#define MRF24J40_PARAM_SPI          (SPI_DEV(0))
#define MRF24J40_PARAM_SPI_CLK      (SPI_CLK_100KHZ)
#define MRF24J40_PARAM_CS           (GPIO_PIN(PA, 24))
#define MRF24J40_PARAM_INT          (GPIO_PIN(PA, 28))
#define MRF24J40_PARAM_RESET        (GPIO_PIN(PA, 19))

#define MRF24J40_PARAMS_BOARD     { .spi = MRF24J40_PARAM_SPI, \
                                      .spi_clk = MRF24J40_PARAM_SPI_CLK, \
                                      .cs_pin = MRF24J40_PARAM_CS, \
                                      .int_pin = MRF24J40_PARAM_INT, \
                                      .reset_pin = MRF24J40_PARAM_RESET }

#define MRF24J40_PARAM_WAKE            (GPIO_PIN(PA, 27))`

And my periph_conf.h concerning the mrf24j40
`
static const spi_conf_t spi_config[] = {
    {   /* EXT2 */
        .dev      = &SERCOM3->SPI,
        .miso_pin = GPIO_PIN(PA, 25), //25
        .mosi_pin = GPIO_PIN(PA, 22), //22
        .clk_pin  = GPIO_PIN(PA, 23),
        .miso_mux = GPIO_MUX_C,
        .mosi_mux = GPIO_MUX_C,
        .clk_mux  = GPIO_MUX_C,
        .miso_pad = SPI_PAD_MISO_3,
        .mosi_pad = SPI_PAD_MOSI_0_SCK_1
    }
};

#define SPI_NUMOF           (sizeof(spi_config) / sizeof(spi_config[0]))`

makefile.deb is configured to import the mrf24j40 module and its include dir

With this configuration (without mrf24j40 autoinit or imported), I can sucessfully initialise and run the bench of test_spi on my spi_dev(0)
However, during the autoinit of the mrf24j40, the spi communication fail between the microcontroller and the mrf24j40 (message issued from mrf24j40_internal.c with ENABLE_DEBUG(1))
The mrf24j40 always respond 0x00, so the write and read tests fail.

What am I doing wrong ?

Thanks for your help

florian-popup, 2017-10-26:
Some update

I checked the SPI communication with an arduino and my board. I can write on the SPI from my board to the Arduino with sucess.

I found out that some of my mrf24j40 had their connection between pad torn up (I had to resolder them because of a poor pad design). I changed all of the one that have been torn up.

I still get the [MRF24J40] Initialisation failure but it now it get 'through' the init process and display some message : interrupt, end IRQ, TX prepare

I will investigate the read capability on the SPI of the microcontroller

bergzand, 2017-10-26:
Hello,

> However, during the autoinit of the mrf24j40, the spi communication fail between the microcontroller and the mrf24j40 (message issued from mrf24j40_internal.c with ENABLE_DEBUG(1))

These problems are the reason why there is code to test the communication between the SPI master and the mrf24j40. :)

> I found out that some of my mrf24j40 had their connection between pad torn up (I had to resolder them because of a poor pad design). I changed all of the one that have been torn up.

That sounds like something that could be an issue. 

The rest of your config looks good to me, I would suggest raising the clock speed of the SPI to at least 1Mhz. Although 100Khz should work just fine, I've had some weird issues when the clock speed is too low.

Do you have a logic analyzer or an oscilloscope available that you could use to verify the SPI communication between the atsamd and the mrf24j40?

florian-popup, 2017-10-26:
So I put the clock speed at 1MHZ and soldered way shorter wires to reach that speed and now it's working like a charm !

Thank you very much !

When you say
> These problems are the reason why there is code to test the communication between the SPI master and the mrf24j40. :)

Where are they ? I just found the periph_spi under tests to test the spi but nothing directly related to mrf24j40 but in mrf24j40_internal.c if ENABLE_DEBUG is enabled

bergzand, 2017-10-26:
Great to hear that!

> Where are they ? I just found the periph_spi under tests to test the spi but nothing directly related to mrf24j40 but in mrf24j40_internal.c if ENABLE_DEBUG is enabled

[Here](https://github.com/RIOT-OS/RIOT/blob/master/drivers/mrf24j40/mrf24j40_internal.c#L44) is the code that verifies the SPI connection between the SPI master and the mrf24j40. It is only included when debug is enabled in `mrf24j40_internal.c`. I've added that code to the driver because it helps to quickly distinguish between connection related issues and driver related issues.

florian-popup, 2017-10-26:
So that's what I used primarily as debug function. It's indeed helpfull
 
Thanks again

closed 2017-10-26 by florian-popup