RIOT-OS/RIOT #7846: stm32f1: I2C read functions return bogus values with unconnected devices

miri64, 2017-10-25:
When testing for the 2017.10 release we noticed that the I2C driver for the stm32f1 returns `>0` when no device is connected (for testing run e.g. `tests/driver_ds1307` on `iotlab-m3` and compare the same application on `samr21-xpro`).

If that issue exists on other STM32 CPUs is not confirmed but needs to be tested as well. That is why I assigned a potpourri of different STM32 contributors to this issue.

PeterKietzmann, 2017-11-27:
Just for reference: #6575 and #6576 might take this into account

tcschmidt, 2018-06-01:
@miri64 will this still be relevant after the I2C refactoring?

aabadie, 2018-06-02:
It depends on the quality of the refactoring for STM32.
If the new API is correctly applied, this should be fixed but I'd prefer waiting for it to be fully done, merged and verified before closing.

cladmi, 2018-08-11:
@aabadie ping as it was listed in https://github.com/RIOT-OS/RIOT/projects/16