RIOT-OS/RIOT #7871: tests: add pexpect scripts to enable `make test`

haukepetersen, 2017-10-26:
Doing the release spec tests, we found that having pexpect scripts for our tests is a good idea, as this is a first step towards automating the tests. I went now through all the tests and found, that there is a large number of them, that can easily be scripted/automated. I think it would make much sense to do so soonish, as this will save us much time down the line!

Trivial to automate:
  [ ] fmt_print
  [ ] irq
  [ ] libfixmath
  [ ] libfixmath_unittests
  [ ] msg_try_receive
  [ ] netdev_test
  [ ] netstats_l2
  [ ] nhdp
  [ ] pipe
  [ ] pkg_jsmn
  [ ] pkg_libcoap
  [ ] pkg_micro-ecc
  [ ] pkg_micro-ecc-with-hwrng
  [ ] pkg_minmea
  [ ] ps_schedstatistics
  [ ] pthread
  [ ] pthread_barrier
  [ ] pthread_cleanup
  [ ] pthread_condition_variable
  [ ] pthread_cooperation
  [ ] pthread_rwlock
  [ ] pthread_tls
  [ ] sched_testing
  [ ] shell
  [ ] sizeof_tcb
  [ ] ssp
  [ ] struct_tm_utility
  [ ] thread_cooperation
  [ ] thread_exit
  [ ] thread_flags
  [ ] thread_flags_xtimer
  [ ] thread_msg
  [ ] thread_msg_block_w_queue
  [ ] thread_msg_block_wo_queue
  [ ] thread_msg_seq

Need some (trivial) modifications before being able to automate:
 - [ ] pkg_u8g2
 - [ ] posix_sleep
 - [ ] trickle

If we work together on this, we should be able to finish this list pretty quickly :-)

aabadie, 2017-10-26:
Nice one!

aabadie, 2017-10-26:
If nobody objects, I can work on the pthread tests

smlng, 2017-10-27:
could everyone adapting a test also check if the list of blacklisted or board with insufficient memory is still up to date. I recently found that it is not, see e.g. #7836 were I reduced the number considerably.

aabadie, 2017-11-15:
All listed tests marked above are now associated to PRs, we are getting closer.

Maybe #8036 could also be of interest for the pexpect scripts.

cladmi, 2017-11-16:
Merged last aabadie PR to add tests.

@haukepetersen could you re-run all tests and validate everything is fixed ?

haukepetersen, 2017-11-16:
why don't you do it yourself: https://github.com/haukepetersen/riotsandbox/blob/master/tools/tester/tester.py

aabadie, 2017-11-16:
> why don't you do it yourself:

interesting and useful script. Why not extending the actual `testrunner` with a `run_all` function ?

miri64, 2017-11-16:
> interesting and useful script. Why not extending the actual testrunner with a run_all function ?

@kaspar030 has some better idea to [improve the testrunner](https://github.com/RIOT-OS/RIOT/pull/7258) we rather should work towards that. ;-)

smlng, 2017-12-01:
closing, all tracked tests are adapted

closed 2017-12-01 by smlng

kaspar030, 2017-12-01:
Good job to everyone involved!