RIOT-OS/RIOT #7875: "Minor" compiling issues found by clang

kYc0o, 2017-10-26:
Here you have a list of compiling issues I found using clang (with its respective PRs):

  - [x] **tests_driver_adcxx1c** fixed in #7856.
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_adcxx1c/main.c:31:28: error: unused parameter 'arg' [-Werror,-Wunused-parameter]
static void alert_cb(void *arg)
                           ^
1 error generated.
```
  - [ ] **tests_driver_at86rf2xx** fixed in #7860
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_at86rf2xx/addr.c:23:23: error: comparison of integers of different signs: 'int' and 'size_t'
      (aka 'unsigned long') [-Werror,-Wsign-compare]
    for (int i = 0; i < addr_len; i++) {
                    ~ ^ ~~~~~~~~
1 error generated.
```
And some more which came after this one was solved.

  - [ ] **tests_driver_enc28j60** fixed in #7857
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/enc28j60/enc28j60.c:259:23: error: comparison of integers of different signs: 'int' and
      'unsigned int' [-Werror,-Wsign-compare]
    for (int i = 0; i < count; i++) {
                    ~ ^ ~~~~~
/Users/facosta/git/RIOT-OS/RIOT/drivers/enc28j60/enc28j60.c:337:19: error: comparison of integers of different signs: 'int' and
      'unsigned int' [-Werror,-Wsign-compare]
        if (res++ >= STARTUP_TIMEOUT) {
            ~~~~~ ^  ~~~~~~~~~~~~~~~
2 errors generated.
```
  - [ ] **tests_driver_encx24j600** fixed in #7858
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/encx24j600/encx24j600.c:304:23: error: comparison of integers of different signs: 'int' and
      'unsigned int' [-Werror,-Wsign-compare]
    for (int i = 0; i < count; i++) {
                    ~ ^ ~~~~~
1 error generated.
```
  - [ ] **tests_driver_isl29125**
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_isl29125/main.c:88:23: error: comparison of integers of different signs: 'int' and
      'unsigned long' [-Werror,-Wsign-compare]
    for (int i = 0; i < sizeof(modes); i++) {
                    ~ ^ ~~~~~~~~~~~~~
1 error generated.
```
  - [ ] **tests_driver_lsm6dsl** I don't have a solution for this, not sure is a problem...
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/lsm6dsl/lsm6dsl.c:72:16: error: binary integer literals are a GNU extension
      [-Werror,-Wgnu-binary-literal]
    if (tmp != LSM6DSL_WHO_AM_I) {
               ^
/Users/facosta/git/RIOT-OS/RIOT/drivers/lsm6dsl/include/lsm6dsl_internal.h:132:46: note: expanded from macro 'LSM6DSL_WHO_AM_I'
#define LSM6DSL_WHO_AM_I                    (0b01101010)
                                             ^
1 error generated.
```
  - [ ] **tests_driver_nrf24l01p_lowlevel** fixed in #7855
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_nrf24l01p_lowlevel/main.c:114:34: error: unused parameter 'arg' [-Werror,-Wunused-parameter]
void *nrf24l01p_rx_handler(void *arg)
                                 ^
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_nrf24l01p_lowlevel/main.c:212:23: error: comparison of integers of different signs: 'int' and
      'unsigned long' [-Werror,-Wsign-compare]
    for (int i = 0; i < sizeof(tx_buf); i++) {
                    ~ ^ ~~~~~~~~~~~~~~
2 errors generated.
```
  - [ ] **tests_driver_pdc8544** fixed in #7859
[... 138 lines cut ...]
    for (int i = 0; i < sizeof(page_mem); i++) {
                    ~ ^ ~~~~~~~~~~~~~~~~
```
I tried to make `size_t i = 0` but then it throws an error on the case which `i` (now signed) cannot be less than 0...

  - [ ] **tests_periph_i2c** didn't fix yet
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/periph_i2c/main.c:184:30: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    if (length < 1 || length > BUFSIZE) {
                      ~~~~~~ ^ ~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/tests/periph_i2c/main.c:232:30: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
    if (length < 1 || length > BUFSIZE) {
                      ~~~~~~ ^ ~~~~~~~
```
  - [ ] **tests_periph_pwm** don't have a fix yet
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/periph_pwm/main.c:71:33: error: comparison of integers of different signs: 'int' and 'unsigned int'
      [-Werror,-Wsign-compare]
        if (state <= 0 || state >= STEPS) {
                          ~~~~~ ^  ~~~~~
```

So far is what I found. This is an effort towards #1121.

aabadie, 2018-04-16:
#7919 was merged so is this issue still valid @kYc0o ?

kYc0o, 2018-04-16:
I'd say all test should be OK right now, but let me check again in case something is missing.

kYc0o, 2018-04-16:
I've found some more. I'll post it here so we still keep track of the current errors, even if they're not critical for the release.

  - [ ] driver_ata8520e **non portable printf arguments**
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_ata8520e/main.c:84:70: error: format specifies type 'int' but the argument has type
      'unsigned long' [-Werror,-Wformat]
                       "message length is %d", SIGFOX_MAX_TX_LENGTH, strlen(argv[2]));
                                          ~~                         ^~~~~~~~~~~~~~~
                                          %lu
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_ata8520e/main.c:103:66: error: format specifies type 'int' but the argument has type
      'unsigned long' [-Werror,-Wformat]
                   "message length is %d", SIGFOX_MAX_TX_LENGTH, strlen(argv[2]));
                                      ~~                         ^~~~~~~~~~~~~~~
                                      %lu
2 errors generated.
```

  - [ ] driver_hd44780 **conficting types**
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/hd44780/hd44780.c:71:13: error: conflicting types for '_send'
static void _send(const hd44780_t *dev, uint8_t value, hd44780_state_t state)
            ^
/Users/facosta/git/RIOT-OS/RIOT/drivers/hd44780/hd44780.c:35:13: note: previous declaration is here
static void _send(const hd44780_t *dev, uint8_t value, uint8_t mode);
            ^
1 error generated.
```

  - [ ] driver_hih6130 **various integer errors**
```console
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_hih6130/main.c:74:42: error: using integer absolute value function 'abs' when argument is of
      floating point type [-Werror,-Wabsolute-value]
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_hih6130/main.c:74:42: note: use function 'fabsf' instead
            (int)integral, (unsigned int)abs(fractional * 10000.f));
                                         ^~~
                                         fabsf
/Users/facosta/git/RIOT-OS/RIOT/tests/driver_hih6130/main.c:77:42: error: using integer absolute value function 'abs' when argument is of
      floating point type [-Werror,-Wabsolute-value]
            (int)integral, (unsigned int)abs(fractional * 10000.f));
```

  - [ ] driver_lis3mdl **implicit conversions**
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/lis3mdl/include/lis3mdl_params.h:75:5: error: implicit conversion from enumeration type
      'lis3mdl_odr_t' to different enumeration type 'lis3mdl_scale_t' [-Werror,-Wenum-conversion]
    LIS3MDL_PARAMS
    ^~~~~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/drivers/lis3mdl/include/lis3mdl_params.h:62:51: note: expanded from macro 'LIS3MDL_PARAMS'
                                       .scale   = LIS3MDL_PARAM_ODR,    \
                                                  ^~~~~~~~~~~~~~~~~
/Users/facosta/git/RIOT-OS/RIOT/drivers/lis3mdl/include/lis3mdl_params.h:47:38: note: expanded from macro 'LIS3MDL_PARAM_ODR'
#define LIS3MDL_PARAM_ODR           (LIS3MDL_ODR_10Hz)
                                     ^~~~~~~~~~~~~~~~
1 error generated.
```

  - [ ] driver_lsm6dsl **binary literals usage** I thought we reached a consensus on the usage of those isn't it?
```console
/Users/facosta/git/RIOT-OS/RIOT/drivers/lsm6dsl/lsm6dsl.c:72:16: error: binary integer literals are a GNU extension
      [-Werror,-Wgnu-binary-literal]
    if (tmp != LSM6DSL_WHO_AM_I) {
               ^
/Users/facosta/git/RIOT-OS/RIOT/drivers/lsm6dsl/include/lsm6dsl_internal.h:132:46: note: expanded from macro 'LSM6DSL_WHO_AM_I'
#define LSM6DSL_WHO_AM_I                    (0b01101010)
                                             ^
1 error generated.
```

[... 159 lines cut ...]
  ...&& (LOGGER_LEVEL > log_lvl)) printf("%lu |  err | %5s (%d)| " msg "\r\n", bsp_getSec(), __FILE__, __LINE__, ##__VA_ARGS__); }while (0)
                                                                                                                 ^
11 errors generated.
```

  - [ ] gnrc_lwmac **comparison errors**
```console
/Users/facosta/git/RIOT-OS/RIOT/sys/net/gnrc/link_layer/lwmac/lwmac.c:225:27: error: comparison of integers of different signs: 'int' and
      'uint32_t' (aka 'unsigned int') [-Werror,-Wsign-compare]
    if (rtt_get_counter() < last) {
        ~~~~~~~~~~~~~~~~~ ^ ~~~~
/Users/facosta/git/RIOT-OS/RIOT/sys/net/gnrc/link_layer/lwmac/lwmac.c:234:17: error: comparison of integers of different signs: 'uint32_t'
      (aka 'unsigned int') and 'int' [-Werror,-Wsign-compare]
    while (last < (rtt_get_counter() + GNRC_LWMAC_RTT_EVENT_MARGIN_TICKS)) {
           ~~~~ ^  ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
/Users/facosta/git/RIOT-OS/RIOT/sys/net/gnrc/link_layer/lwmac/rx_state_machine.c:211:55: error: format specifies type 'void *' but the
      argument has type 'gnrc_pktsnip_t *' (aka 'struct gnrc_pktsnip *') [-Werror,-Wformat-pedantic]
        LOG_DEBUG("[LWMAC-rx] Inspecting pkt @ %p\n", pkt);
/Users/facosta/git/RIOT-OS/RIOT/sys/net/gnrc/link_layer/lwmac/lwmac.c:801:91: error: format specifies type 'void *' but the argument has
      type 'gnrc_pktsnip_t *' (aka 'struct gnrc_pktsnip *') [-Werror,-Wformat-pedantic]
                    LOG_ERROR("ERROR: [LWMAC] Can't push RX packet @ %p, memory full?\n", pkt);
```

That's what I found so far. Though, I'm not sure if all of them are true positives, but we might want to address them at some point.

aabadie, 2018-09-20:
@kYc0o, if I'm not mistaken, Clang is now part of the CI (for native at least). So all issues here should be fixed now, right ?

kYc0o, 2018-09-21:
I'll test again this compile issues to be completely sure they're gone before closing.

miri64, 2018-09-21:
@kYc0o maybe also have a look at #9398

aabadie, 2018-10-01:
#9809 was merged so I think we can close this one. @kYc0o did you re-test that the issues are gone ?

aabadie, 2018-10-19:
ping @kYc0o

closed 2019-09-10 by stale[bot]