RIOT-OS/RIOT #7877: SPI connection cant read 16 bit side register.

maitysubhasis, 2017-10-26:
When I am trying to read a 16 bit register for MAC address from ENC28J60 only first 8 bit is read properly.
The second bit is always read as zero.
[image]
Here first line is CLOCK and second line is MISO. Data is read in rising edge. The output should be 01110011 11001110 but I get 01110011 00000000. I am using CC2538 microcontroller.

gebart, 2017-10-26:
You need to provide a MWE (minimal working example) or at least some code to show how you are reading the SPI bus.

maitysubhasis, 2017-10-26:
@gebart I am calling  `spi_transfer_reg(spi_t bus, spi_cs_t cs, uint8_t reg, uint8_t out)` from $RIOT/drivers/spic.c  
```c
void spi_transfer_regs(spi_t bus, spi_cs_t cs,
                       uint8_t reg, const void *out, void *in, size_t len)
{
    spi_transfer_bytes(bus, cs, true, &reg, NULL, 1); // su: send command
    spi_transfer_bytes(bus, cs, false, out, in, len); // su: read incoming data
}
```
`spi_transfer_bytes(...)` is in 
$RIOT/cpu/cc2538/periph/spi.c
```c
void spi_transfer_bytes(spi_t bus, spi_cs_t cs, bool cont,
                        const void *out, void *in, size_t len)
{
    const uint8_t *out_buf = out;
    uint8_t *in_buf = in;

    assert(out_buf || in_buf);

    if (cs != SPI_CS_UNDEF) {
        gpio_clear((gpio_t)cs); // su: clear chip select
    }

    // Transmit the data
    if (!in_buf) {
        for (size_t i = 0; i < len; i++) {
            while (!(dev(bus)->SR & SSI_SR_TNF)) {}
            dev(bus)->DR = out_buf[i]; // su: writes to Transmit FIFO
        }
        /* flush RX FIFO while busy*/
        while ((dev(bus)->SR & SSI_SR_BSY)) {
            dev(bus)->DR; // subhasis: clear Receive FIFO
        }
    }
    // Receive the data
    else if (!out_buf) { /*TODO this case is currently untested */
        size_t in_cnt = 0;
        for (size_t i = 0; i < len; i++) {
            while (!(dev(bus)->SR & SSI_SR_TNF)) {}
            dev(bus)->DR = 0; // put 0 in Transmit FIFO
            if (dev(bus)->SR & SSI_SR_RNE) {
                in_buf[in_cnt++] = dev(bus)->DR;
            }
        }
        /* get remaining bytes */
        while (dev(bus)->SR & SSI_SR_RNE) {
            in_buf[in_cnt++] = dev(bus)->DR;
        }
    }
    // first Transmit the data then Receive the data
    else {
        for (size_t i = 0; i < len; i++) {
            while (!(dev(bus)->SR & SSI_SR_TNF)) {}
            dev(bus)->DR = out_buf[i];
            while (!(dev(bus)->SR & SSI_SR_RNE)){}
            in_buf[i] = dev(bus)->DR;
        }
    /* wait until no more busy */
    while ((dev(bus)->SR & SSI_SR_BSY)) {}
    }

    if ((!cont) && (cs != SPI_CS_UNDEF)) {
        gpio_set((gpio_t)cs);
    }
}
```

gebart, 2017-10-27:
@maitysubhasis I meant that we need to see your own code, the RIOT code is already available here on GitHub. Where do you get zeroes returned? Is the enc28j60 driver reading the MAC address as 00:00:00:00:00:00?

maitysubhasis, 2017-10-27:
Yes the driver is reading the address as 00:00:00:00:00:00(when creating frames).

closed 2018-11-28 by maitysubhasis