RIOT-OS/RIOT #7888: Low-level Hardware Support

kYc0o, 2017-10-27:
Improved MIPS support
  - Related issues
    - [ ]

Radio support for TI SensorTag
  - Related issues
    - [ ]

Radio support for Silab Thunderboard
  - Related issues
    - [ ]

Improve x86/Quark support (e.g. Arduino 101) via QMSI
  - Related issues
    - [ ]

Radio support for Arduino MKR1000
  - Related issues
    - [ ]

ST Nucleo support
  - Related issues
    - [ ]

LoRa drivers
  - Related issues
    - [ ]

kbumsik, 2017-10-27:
FYI, [Intel discontinued the whole IoT products](https://www.cnx-software.com/2017/07/26/intel-curie-module-arduino-101-boards-are-being-discontinued-too/), including Curie, Quark, and Arduino 101, meaning Intel left the IoT devices market and not willing to release new products anymore.

kYc0o, 2017-10-29:
Yeah I was aware of that so I suppose we need to erase that from our roadmap.

PeterKietzmann, 2019-02-21:
This issue was never used or maintained, so IMO it didn't help tracking development process. I guess there are others of this type. @emmanuelsearch how do you think about it? Should we update or close? Personally I think that especially hardware support is so fast that tracking it would only introduce overhead.

closed 2019-09-25 by stale[bot]

kaspar030, 2019-09-25:
IMO ok to close.