RIOT-OS/RIOT #7890: Software Updates

kYc0o, 2017-10-27:
Prototype of firmware swapping, assuming the internal flash can contain 2 images and the bootloader, demoed on IoT-Lab_M3
  - Related issues
    - [ ]

Firmware loading through serial using ethos, when no image is flashed on the node.
  - A firmware server can be started with the address and the firmware to flash.
  - Related issues
    - [ ]

Firmware update through UDP.
  - A module is included on any updatable firmware, able to deal with firmware requests.
    - Related PRs
      - [ ]

  - The node asks for a new firmware and the server answers with the correct image.
    - Related PRs
      - [ ]

Interpreted language support (js and python) with dynamically updatable scripts
  - Related PRs
    - [ ]

Prototype for secure bootloading enhancements (authentification, integrity, confidentiality, bootstrap of keys)
  - Related PRs
    - [ ]

Firmware distribution enhancements (firmware metadata refinements, automatic discovery/push/pull from server or nearby node caching newer firmware)
  - Related PRs
    - [ ]

Partial firmware update (Dynamic linking and loading of ELF files).
  - Related PRs
    - [ ]

waehlisch, 2018-09-21:
Is this issue still needed or can we close because of #9342.

danpetry, 2018-09-21:
All of the linked issues are closed, apart from micropython (which currently has tenuous link to software updates) and MCUBoot support for FRDM-K64F, which is linked on the OTA project (here)[https://github.com/RIOT-OS/RIOT/projects/11]. Closed.

closed 2018-09-21 by danpetry