RIOT-OS/RIOT #7893: Network Stack Low layers

kYc0o, 2017-10-27:
BLE link-layer support (below HCI, port from MyNewt?) towards fully open source support of 6LoWPAN over BLE
  - Related issues
    - [ ]

LoRa & LoRaWAN driver and MAC layer integration
  - Related issues
    - [ ]

Duty Cycling Radio: finalize and merge lwMAC
  - Related issues
    - [ ]

Traffic adaptive MAC protocol: finalize and merge iQueue
  - Related issues
    - [ ]

Retransmissions by MAC: finalize and merge netdev_retrans
  - Related issues
    - [ ]

Point-to-Point Protocol (PPP): finalize and merge gnrc_ppp
  - Related issues
    - [ ]

Re-integrate and clean-up 802.15.4 TSCH integration
  - Related issues
    - [ ]

dkazantzas, 2017-10-28:
Hi, I would love to help with that. How could I acccomplish that?

kYc0o, 2017-11-03:
@dkazantzas we started a thread on our mailing list to discuss about this topic. You can subscribe at [devel@riot-os.org](https://lists.riot-os.org/mailman/listinfo/devel).

PeterKietzmann, 2018-06-26:
@kYc0o personally I have the impression this issue does not help. And obviously it hasn't been updated or used since opening. Browsing through the referenced PRs/issues it seems that (i) many projects have been canceled and closed and (ii) I'm missing the red thread. Of course it's about "lower layers" but for example, one finds further tracking issues for MCU hardware support next to outdated radio driver PRs or protocol feature requests. What do you think about this issue, candidate for closing or would you mind updating?

kYc0o, 2018-06-26:
@PeterKietzmann I agree with you. My aim was to track the progress of the things highlighted on our roadmap, in order to know where the efforts were needed. I don't know if we're actually following the roadmap anymore of if it's still up to date. I can try to update all my issues and (try to) give again a global view of the progress made in our most important directions. But if the need doesn't exist, or needs to be tackled differently, I can close all my issues.

PeterKietzmann, 2018-06-26:
> My aim was to track the progress of the things highlighted on our roadmap

Yepp I see and the idea seems reasonable. But now -half a year later - it seems as if the tracking mechanism was not used by the community and thus only introduces management overhead. However, I won't close the issue, just wanted to raise the question...

closed 2019-09-10 by stale[bot]