RIOT-OS/RIOT #7894: Network Stack High layers

kYc0o, 2017-10-27:
Fix lingering NDP issues
  - Related issues
    - [ ]

Finalize and merge sock API (still missing: emb6), remove conn API
  - Related issues
    - [ ]

ICN stack support clean-up
  - Related issues
    - [ ]

Discuss mid- and long-term plans for network stack maintenance & development (GNRC vs other supported stacks)
  - Related issues
    - [ ]

Support for CoAP blockwise transfer, observe
  - Related issues
    - [ ]

Revisit network time synchronization
  - Related issues
    - [ ]

miri64, 2017-10-27:
What is this issue? Is this a copy of the Roadmap? I'm confused :-/

kYc0o, 2017-10-27:
Yes, it's a step forward the organisation for the meetings.

miri64, 2017-10-27:
I'm not sure if having 9 issues for 1 meeting is scaling so well...

miri64, 2017-10-27:
One point is actually outdated (sorry for not updating the Roadmap): conn was deleted in April.

miri64, 2017-10-27:
(adapted the issue and the roadmap)

kYc0o, 2017-10-27:
> I'm not sure if having 9 issues for 1 meeting is scaling so well...

No that's not the point,  I'll send an email to explain.

tcschmidt, 2018-06-21:
I guess we can close this one - correct?

kYc0o, 2018-06-26:
I'm not sure, I need to check if all issues are solved, though I guess there are more... I need rather to update all the related issues to reflect the ongoing work.

kb2ma, 2018-06-26:
I maintain some pages for CoAP, which are referenced from the overall [Roadmap page](https://github.com/RIOT-OS/RIOT/wiki/Roadmap). #9309 is a pretty good summary of current state.

closed 2019-09-10 by stale[bot]