RIOT-OS/RIOT #7909: gnrc_networking runs fail on raspberry pi 3

florian-popup, 2017-10-30:
Hello,

I try to run the gnrc_networking example on my rpi 3 under ubuntu mate, however it fails running
I followed the [Instructions](https://github.com/RIOT-OS/RIOT/tree/master/examples/gnrc_networking)
-netcat-openbsd is installed from the ubuntu mate repository
I ran 
`sudo modprobe ipv6
sudo ip tuntap add tap0 mode tap user ${USER}
sudo ip link set tap0 up
sudo make
sudo make term` 

> pi-popup@pi-popup:~/prog/origin/RIOT/examples/gnrc_networking$`
 sudo make term
/home/pi-popup/prog/origin/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

> main(): This is RIOT! (Version: 2018.01-devel-134-gb80cd-pi-popup)
RIOT network stack example application
All up, running the shell now
> /home/pi-popup/prog/origin/RIOT/examples/gnrc_networking/../../Makefile.include:386: recipe for target 'term' failed
make: *** [term] Illegal instruction

Same thing happened with PORT=tap0 make term
BUT gnrc_minimal runs perfectly

Any help on how to debug that ?
Thanks

miri64, 2017-11-01:
Hi @florian-popup,

did you call the `dist/tools/tapsetup/tapsetup` script beforehand? In general I don't exactly know, if the `gnrc_border_router` example currently runs sensible on native, since we only have support for Ethernet there at the moment.

Cheers,
Martine

miri64, 2017-11-01:
@cgundogan did you saw something similar on your Pi3 setup? (also mentioning you because of the RPL issue)

cgundogan, 2017-11-01:
On my pi3 it works with current master:
```
pi@pi3:~/RIOT/examples/gnrc_networking $ make clean all term
...
   text    data     bss     dec     hex filename
 205129     604  105316  311049   4bf09 /home/pi/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf
/home/pi/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.01-devel-154-gcb9b6-pi3)
RIOT network stack example application
All up, running the shell now
> 
```

@florian-popup  What gcc version do you use?
```
pi@pi3:~/RIOT/examples/gnrc_networking $ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Could use also provide more info on your architecture? (32-bit, 64-bit)?
```
pi@pi3:~/RIOT/examples/gnrc_networking $ uname -a
Linux i3-gateway 4.9.41-v7+ #1023 SMP Tue Aug 8 16:00:15 BST 2017 armv7l GNU/Linux
```

On my pi3 I use:
```
pi@pi3:~/RIOT/examples/gnrc_networking $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 9.1 (stretch)
Release:        9.1
Codename:       stretch
```

florian-popup, 2017-11-02:
@miri64 Thanks for your help

Usually I bring the tap0 interface myself as in the gnrc_networking example 
sudo ip tuntap add tap0 mode tap user ${USER}
sudo ip link set tap0 up

I tried to do it via the tapsetup script but I got the exact same error when executing gnrc_networking.
gnrc_border_router is a bit different for me because I need to modify the code, I will use directly an interface instead of the Slip. I'm trying to focus on making gnrc_networking working yet

@cgundogan Thanks for your help too

I had a gcc version of 5.somthing. I updated it to the 6.3 version and now I get a different error message

> pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ sudo make clean all term
...
 text	  data	   bss	   dec	   hex	filename
 141323	   592	105324	247239	 3c5c7	/opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf
/opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0 
RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

> main(): This is RIOT! (Version: 2018.01-devel-156-g9b32d-pi-popup)
RIOT network stack example application
All up, running the shell now
> /opt/src/RIOT/examples/gnrc_networking/../../Makefile.include:386: recipe for target 'term' failed
make: *** [term] Segmentation fault

Compiled & executed directly after a fresh git clone and after executer the dist/tools/tapsetup/tapsetup script

Here is my config after I upgraded the kernel via rpi-update and installed the 6.3 version of gcc

Arm-none-eabi-gcc

> pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

gcc

pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ gcc --version
gcc (Ubuntu/Linaro 6.3.0-18ubuntu2~16.04) 6.3.0 20170519
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE

uname -a

> pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ uname -a
Linux pi-popup 4.9.59-v7+ #1047 SMP Sun Oct 29 12:19:23 GMT 2017 armv7l armv7l armv7l GNU/Linux

lsb release

> pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.3 LTS
Release:	16.04
Codename:	xenial

cgundogan, 2017-11-02:
>Notice the Segmentation fault instead of the Illegal instruction (Maybe it's just the compiler that changed the message)

well, we can work with seg faults. Illegal instructions are much more complicated (:

Could you try to pin-point where the segfault is thrown?

Try:
```sh
make clean all-debug debug
```
and then `run` in the gdb shell.

florian-popup, 2017-11-03:
Oh, I was wondering why I didn't get any more explicit error message :-)

Here is the output

pi-popup@pi-popup:/opt/src/RIOT/examples/gnrc_networking$ sudo make clean all term
..
   text	  data	   bss	   dec	   hex	filename
 141323	   592	105324	247239	 3c5c7	/opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf
gdb -q --args /opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0 
Reading symbols from /opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf...done.
(gdb) run
Starting program: /opt/src/RIOT/examples/gnrc_networking/bin/native/gnrc_networking.elf tap0
Cannot parse expression `.L1185 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.

RIOT native interrupts/signals initialized.
LED_RED_OFF
LED_GREEN_ON
RIOT native board initialized.
RIOT native hardware initialization complete.

main(): This is RIOT! (Version: 2018.01-devel-197-gd9bb0-pi-popup)
RIOT network stack example application
All up, running the shell now
> 
Program received signal SIGSEGV, Segmentation fault.
0x00014530 in native_interrupt_init ()
    at /opt/src/RIOT/cpu/native/irq_cpu.c:539
539	       err(EXIT_FAILURE, "native_interrupt_init: sigdelset");
(gdb)

cgundogan, 2017-11-03:
@florian-popup hm that's weird. I have no idea why registering SIGINT should fail. Another thing I notice is that your gdb says:
```
Cannot parse expression `.L1185 4@r4'.
warning: Probes-based dynamic linker interface failed.
Reverting to original interface.
```

I don't see that on my pi3 .. could this problem be related to that? Some (quick) googling didn't reveal the solution, but this **warning** of gdb seems to be known on arm architectures.

florian-popup, 2017-11-03:
https://bugzilla.redhat.com/show_bug.cgi?id=119618

Speaks about this one exactly

florian-popup, 2017-11-06:
@cgundogan @miri64 
I guess that the errors I get are related to the OS.
I don't get any errors from running it under raspbian but I get it when running under ubuntu mate.
It can related to the same error that makes the newer version of firefox crash on ubuntu mate in raspberry pi

cgundogan, 2017-11-06:
@florian-popup thanks for the hint and testing with raspbian and ubuntu. Do you have any clue whether they use different libc versions?

florian-popup, 2017-11-06:
So far my setup is

Raspbian (gnrc_networking working), rpi3

> ldd --version
ldd (Debian GLIBC 2.24-11+deb9u1) 2.24
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

Ubuntu Desktop (gnrc_networking working), pc, intel 64 bits

> ldd --version
ldd (Ubuntu GLIBC 2.23-0ubuntu9) 2.23
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

Ubuntu Mate (gnrc_networking not working), rpi3

> ldd --version
ldd (Ubuntu GLIBC 2.23-0ubuntu9) 2.23
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

florian-popup, 2017-11-06:
@cgundogan 

I installed the libc6 version of stretch (same as Raspbian) directly from the stretch repo on my ubuntu mate
So now I get 

> ldd --version
ldd (Debian GLIBC 2.24-11+deb9u1) 2.24
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Written by Roland McGrath and Ulrich Drepper.

But the error is still here when I execute gnrc_networking

PeterKietzmann, 2019-02-21:
@florian-popup did this problem ever resolve?

closed 2019-09-25 by stale[bot]