RIOT-OS/RIOT #7915: Question: How does one use gnrc_border_router in a native linux environment?

Stupid-Git, 2017-10-31:
I would like to use gnrc_border_router with a number of instances of gnrc_networking where the wireless interfaces connect to tap0, tap1, tap2, ... created with "tapsetup -c n" and the border interface connected to another interface e.g. "taptowan".   
I would envisage that the boarder router would be invoked in the following manner   
$ bin/native/gnrc_border_router.elf tap0 taptowan   
     
1. Is there a simple way of achieving this?   
2. If someone has a desvirt configuration which includes the use of a border router?   

Many thanks.
Karel

miri64, 2017-11-10:
Simple answer to the top question (at least for the moment): not at all.

There is a PR (#6121) that would allow for that, by providing a [ZEP](https://github.com/boundary/wireshark/blob/master/epan/dissectors/packet-zep.c#L22-L43)-based network device that would allow to either let two or more native platforms communicate or to hand that over to a native WPAN devices. A desvirt configuration for that does not exist, there is however a [prototype of mine to create a simulated network using ZEP](https://github.com/miri64/pywpan).

madokapeng, 2018-04-15:
i am been assigned to this PR. is possible to provide some instruction how? what is the status of current issue?.
many thanks

PeterKietzmann, 2019-02-21:
@Stupid-Git  was there any progress since Oct '17? Did you make use of #6121?

Stupid-Git, 2019-02-26:
Dear @PeterKietzmann, thanks for the inquiry. I have to admit a lot of time has passed since I looked at this issue/feature. I do remember looking at the ZEP but not attempting to use it, because my (sledge hammer) modifications to the source code were giving me the functionality I needed at the time. 
    In general however, I am a firm believer that the software framework should allow as much abstraction (read serialization) as possible, so that the partition between "simulated"/host side and device/radio/phy side can be assigned with a simple build switch. What today will only fit on the host will fit on the device side tomorrow. 
    Sorry I could not give any concrete feedback. Thanks again. Karel

miri64, 2019-08-30:
I guess what @Stupid-Git wanted was #9942. However, as the question was asked years ago and he already states he found another hacky solution, I'd say let's close. Reopen, if you disagree.

closed 2019-08-30 by miri64