RIOT-OS/RIOT #7918: Usage of GCC extension for binary constants

smlng, 2017-10-31:
I was trying to compile RIOT with `-Wpedantic` (+ `-Wall -Werror ...`) and ran into several errors like the following:

```
/RIOT/boards/pba-d-01-kw2x/include/periph_conf.h:52:18: error: binary constants are a GCC extension
     .fll_frdiv = 0b010, /* Divide by 128 */
                  ^
```

It seems that usage of `0b*` is not generally recommended. There are not that many places where such is used in RIOT, however, to get RIOT compiled with `-Wpedantic` in the (far) future we need to find a solution - or wait that compilers don't see this as a problem.

What's the general opinion on this?

bergzand, 2017-10-31:
I personally don't see a lot of extra value in binary constants above hex constants.
My preference would be to have them replaced by their hex equivalent and adhere a bit more to the C standard instead of relying on GCC extensions.

smlng, 2017-10-31:
> don't see a lot of extra value in binary constants above hex constants

@bergzand that's exactly what I was thinking, too - but I want to open the discussion first, before changing anything blindly.

gebart, 2017-10-31:
I like the binary literals because then the raw register values (like the frdiv option) matches literally the numbers in the reference manual and no manual conversion is necessary. Any kind of human conversion is always going to have a risk of introducing errors, and eliminating the extra conversion between hex/dec and binary is one way to reduce the risk. I know these three bit numbers are quite trivial to convert, but 4,5,6 bit fields may be easier to read written in binary than when written in hex.
Anything above 8 bits is not worth writing in binary because it just makes it harder to read..

kaspar030, 2017-11-01:
> I like the binary literals because then the raw register values (like the frdiv option) matches literally the numbers in the reference manual and no manual conversion is necessary.

I think we agree that binary literals do increase readability. But there are reasons for not using gcc extensions (which we can reconsider any time...).

Maybe we can mitigate the issue? Which editor are you using? I've made good experiences with [radical](https://github.com/glts/vim-radical.git) for vim.

closed 2019-09-10 by stale[bot]

kaspar030, 2019-09-10:
I'm not aware of any issue so far that was caused by the current usage of that extension.
I suggest we conclude that for new code they SHOULD be avoided, and we fix existing usage as soon as it gets in the way.