RIOT-OS/RIOT #7941: periph: remove dev_enums.h

haukepetersen, 2017-11-06:
The `drivers/include/periph/dev_enums.h` file was introduced for backwards compatibility when we changed the style of doing the board configuration in `periph_conf.h` some time ago. Most of the periph drivers are remodeled by now, though some are open. Except the `i2c` drivers, which will be remodeled while reworking the complete interface, the following drivers need adaption:

timer:
- [ ] samd21
- [ ] saml21 (best to unify these two into sam0_common in the same process)
- [ ] lm4f120
- [ ] lpc1768

uart:
- [ ] cc2538
- [ ] lm4f120
- [ ] lpc1768

I did not really check the open PRs though, so there might be a fix for some of the mentioned in the pipeline already. Please check before starting to work on any of these...

danpetry, 2017-11-22:
No fixes in open PRs that I can see. Related open PRs are as follows:

- https://github.com/RIOT-OS/RIOT/pull/8058 - some info on sam0_common
- https://github.com/RIOT-OS/RIOT/pull/7694 - affects lm4f120 timer driver
- https://github.com/RIOT-OS/RIOT/pull/7310 - affects cc2538 UART driver

danpetry, 2017-12-01:
I think I also need to change the periph_conf files for the boards that have the above cpus on them. Will include the changes for those in the PR I submit for each cpu peripheral.

haukepetersen, 2017-12-01:
> I think I also need to change the periph_conf files for the boards that have the above cpus on them

Yes, you will certainly :-)

danpetry, 2017-12-13:
@haukepetersen there are aliases for UART_X called UART_X_DEV in the cc2538 driver, I was just wondering what they do?

danpetry, 2018-01-04:
@smlng for the 2538, are the above boards the only ones with this CPU? Also, would you mind testing the changes for the other boards once I've made them? (I only have the openmote at the moment).

danpetry, 2018-01-08:
Note for reviewer: now that I've removed the switch/case statement, device numbers outside the ones actually implemented can be used. They won't be caught by the default statement as before, and there's no value restriction on uart_t. What would be the effect of this?

MrKevinWeiss, 2018-03-21:
Just FYI I am working on updating the uart config for the lm4f120 based off @danpetry work.

MrKevinWeiss, 2018-03-23:
I think I will work on the UART for the lpc1768 next.  There are some shared files that #6472 is working on but it seems to be in different sections.

miri64, 2020-07-01:
@haukepetersen @MrKevinWeiss what is the state on this?

MrKevinWeiss, 2020-07-01:
Ahh yes, one of my first riot tasks still not done :)

MrKevinWeiss, 2020-07-01:
It seems like the `slwstk6220a` board uses I2C0

The `lm4f120` and `lpc1768` use TIMER_x and UART_x.

Both of those boards are obsolete. Maybe we can just hardcode UART_0 for the lm4f120 as the `uart_init` asserts the uart==0 anyways.  Then it will be similar to the nrf52dk I think.

The lpc might take a bit more.

haukepetersen, 2020-07-09:
what he said :-)

MrKevinWeiss, 2020-07-09:
This is on the list, as soon as I put out all the fires I will get to it... or maybe find a student :imp:

PR #15068 (2020-09-25): drivers/dev_enums: Remove unused dev_enums.h
### Contribution description

Now that all dependencies of `dev_enums.h` are resolved we can completely remove it.

### Testing procedure

Only needs a green from the build test.

### Issues/PRs references

Fixes #7941
closed 2020-09-25 by aabadie