RIOT-OS/RIOT #7944: SAML/D/R need more configuration flexibility

travisgriggs, 2017-11-06:
I believe that the configuration options for the SAM chips must change if RIOT on these chips is to be competitive in the realm of low power consumption. The following is an excerpt from an email I originally sent to one of the mailing lists:

> The original samd21 port was basically done by cloning the samr21 and trimming back the radio parts. The samd21 chip is a pretty versatile chip. It sports 5 oscillator sources. Two pseudo sources (1 PLL and 1 DFLL). 9 generic clock generators can be driven from these 7 sources, with a variety of options for configuring dividers and power modes. Configuring your CPU clock and peripherals is a lot like driving a multi transmission semi tractor trailer. You’ve got gearboxes galore to shift around to produce different results.
> 
> The original samr21 offered users a single clock/peripheral configuration using the startup 8MHz RC Oscillator. xtimer was hard coded. Later someone added a second path to use the PLL at higher frequencies. Dan and I have floated PRs to include a DFLL based option. Recently, another configuration PR has been put up. The basic approach to all of these has been to come up with a “recipe” for how to configure the registers, give it a name that is derived from one of the components, try to hunt down all of the existent dependencies and capture them, and use #ifdef branches to capture that particular recipe. It’s like getting a box of legos, but being told you can only put them together 4 ways. And each new recipe has to be expressed as a derivative of the original recipes.
> 
> It is my belief that this is just not a scalable approach to configuring the samd21 for RIOT. It will continue to limit the application programmer's ability to tune the chip's flexibility to their application. I believe that a model that allows the user to better follow the model presented in the data sheets is what is needed. I would expect to be able to configure all of oscillators in one place, preferably in a coding construct/format that approximated something tabular. I would expect to be able to do something with the array of clock generators, indicating which were enabled, sourced to what, and divided appropriately. And I would like to be able to configure each peripheral’s clock source.

I don't have an entirely clear design proposal on how to achieve this yet. But I can see some of the pieces. One of the first is to modify all of the sam0 peripheral drivers as has been done with uart driver, so that their clock source can be configured. A second is to add specifiers to all of the same peripherals to specify the power/standby behavior. In both cases, this includes things like xtimer (or at least, it's core timers), rtt, etc.

It is less clear how to solve the cpu base clock. This is not just a "declarative" thing. Code has to actually be executed to transition from the default 8MHz RC Oscillator, and that path varies depending on where it ends up at. I've wondered if a "pluggable" clock_startup() function would be worthwhile there.

photonthunder, 2017-11-06:
PR #7315 was an attempt to try and address the symptom but does not address the issue.  I do like the idea of a pluggable clock_startup function (default exists but can be overwritten) that is setup with the freedom to set your clock specific to your needs, especially if paired with a peripheral configuration that sets which clock is in control.  Looking at other boards it appears that the STM32 has the same problem where the STM32F family calls stm32_common for its clock configuration (with a ton of if defines).  The stm32L family all setup version specific clock code.  So a RIOT based solution that benefits the sam0 family would also benefit the stm32 family.

gebart, 2017-11-06:
Regarding core clock source: the Kinetis driver uses a (somewhat bloated) configuration struct that is usually placed in ROM, and the periph driver for the MCG (Multipurpose Clock Generator) takes care of starting the relevant oscillator sources depending on which mode was chosen and also provides a function for moving between the different modes to select the chosen FLL/PLL settings (or no FLL/PLL). It is not possible for the hardware to go directly between some of the modes and the driver will need to step through multiple modes until it reaches the target, which seems similar to what you describe of your problems with the RC oscillator on SAMD21. See the figure in the API docs at http://api.riot-os.org/group__cpu__kinetis__common__mcg.html.

The way we solved it for Kinetis was to define a precomputed pathfinding map for stepping from mode X to mode Y, and looping until we reach the target mode see https://github.com/RIOT-OS/RIOT/blob/master/cpu/kinetis_common/periph/mcg.c#L28 (that map is manually constructed based on the information found in the mode figure in the API docs)

The kinetis_mcg_init function is called by the kinetis_common cpu_init function, so it is shared between all boards using these CPUs. By placing the configuration in a struct instead of defining preprocessor macros we gain some error checking from the compiler. This is useful when code is refactored in the future, because it makes sure that no old unused configuration options are left in the board configuration headers, and also gives us some limited type checking. (It has often happened before that configuration macros are left in the periph_conf.h files even after drivers are refactored to use different sets of options)

photonthunder, 2017-11-06:
Looking closer at the stm32 it looks like there is c code that creates a clk configuration that you can then put in your periph_conf.h file.

closed 2019-09-10 by stale[bot]