RIOT-OS/RIOT #7946: sdcard_spi & xtimer issue

fjmolinas, 2017-11-06:
Hi,

I have been using the sdcard_spi test found in RIOT/test/driver_sdcard_spi. I'm running it on master on the nucleo-l1. The test freezes up while initializing the sdcar_spi driver. This happens around this block of code (at sdcard_spi.c):

`                gpio_set(card->params.clk);
                xtimer_usleep(SD_CARD_PREINIT_CLOCK_PERIOD_US/2);
                gpio_clear(card->params.clk);
                xtimer_usleep(SD_CARD_PREINIT_CLOCK_PERIOD_US/2);`

I have the impression it has to do with the small time it's trying to sleep (5us) and the fact that xtimer_backoff for this board is 3u. I have seen that increasing the backoff or increasing the time it sleeps between gpio_clear and gpio_set fixes the problem.

What I'm wandering if the backoff is appropriated or if one shouldn't expect xtimer to work for small sleeps like that, in witch case maybe the driver sleep is too demanding. I tried running the debugger and found it got stuck at gpio_clear with the defualt time settings. Cheers.

gebart, 2017-11-06:
A backoff value of 3 seem low. It's that board running with a 1 MHz hardware timer or is it lower?
The backoff value is used internally by xtimer to avoid setting a timer target in the past and the value is hardware dependent. Try increasing the xtimer backoff parameters, or just remove the definitions and use the defaults provided by xtimer.h

MichelRottleuthner, 2017-11-07:
Related: #7347, #7349
You can use the test `xtimer_usleep_short` to verify if your updated [XTIMER_BACKOFF](https://github.com/RIOT-OS/RIOT/blob/master/boards/nucleo-l152/include/board.h#L35) is high enough.

fjmolinas, 2017-11-08:
Yes any of those fixes the issue. Should I close since its related to  #7347 and #7349?

MichelRottleuthner, 2017-11-08:
Closed because the above mentioned issue already covers this problem.

closed 2017-11-08 by MichelRottleuthner