RIOT-OS/RIOT #7962: When to use FEATURES_PROVIDED vs USEMODULE

photonthunder, 2017-11-08:
I think PR #7241 tries to fix how you add in peripherals but I am still unclear on the proper approach.  For my current board running the latest master (710771a) I need to have most of my periph's declared using USEMODULE:

```
USEMODULE += periph_adc
USEMODULE += periph_cpuid
USEMODULE += periph_flashpage
USEMODULE += periph_gpio
USEMODULE += periph_i2c
USEMODULE += periph_pwm
USEMODULE += periph_rtt
USEMODULE += periph_spi
USEMODULE += periph_timer
```

However, I still have to use FEATURES_PROVIDED for periph_uart:

```
FEATURES_PROVIDED += periph_uart
```

So what is the proper way to include peripherals, and why the discrepancy with periph_uart?

kaspar030, 2017-11-08:
> I need to have most of my periph's declared using USEMODULE

Why do you have to do that?

> So what is the proper way to include peripherals

The proper way is to create proper dependencies. For drivers, it would be ```FEATURES_REQUIRED += periph_xxx```, in ```drivers/Makefile.dep```. If e.g., spi depends on gpio, it needs to be specified in ```cpu/$(CPU)/Makefile.dep```.

Only if a specific port *always* needs code of one of the periphs, it is OK to just add the corresponding USEMODULE line into e.g., the CPU's ```Makefile.include```.

photonthunder, 2017-11-08:
Previously I had everything defined as a feature required in my makefile.features for my board.  That no longer compiles, so I had to include USEMODULE instead to get it to compile.  Except for UART which wouldn't compile unless I used FEATURES_REQUIRED still.

photonthunder, 2017-11-08:
So to restate my question. If I want for example to include the ADC peripheral in my board code what is the RIOT way to do that?  I am assuming USEMODULE is the proper way now, but why does uart still require FEATURES_REQUIRED.  By the way this is the error I get:

```
There are unsatisfied feature requirements: periph_uart

EXPECT ERRORS!
```

kbumsik, 2017-11-09:
Which board are you using? AFAIK `FEATURES_PROVIDED += periph_uart` is not supposed to be used in project Makefiles. `FEATURES_PROVIDED` is used in board [Makefile.features](https://github.com/RIOT-OS/RIOT/blob/master/boards/samr21-xpro/Makefile.features) and `FEATURES_PROVIDED += periph_uart` means, in Makefile.features, the corresponding board has driver port for UART. 

So if you have to use `FEATURES_PROVIDED += periph_uart` in your project Makefile and the UART works correctly, it is a bug so we need to add it in the corresponding Mkaefile.features.

photonthunder, 2017-11-09:
I use the samd21, xplained and then a similar custom board.  Previously, I just had FEATURES_PROVIDED in Makefile.features for the board and everything would compile (because everything got compiled).  With the latest code I added USEMODULE in my Makefile.include for the peripherals I use in order to get it to compile.  What is interesting is the FEATURES_PROVIDED list is irrelevant except for uart.

kaspar030, 2018-02-06:
> What is interesting is the FEATURES_PROVIDED list is irrelevant except for uart.

That is expected, FEATURES_PROVIDED merely states what is available.

Everything in FEATURES_REQUIRED that is not in FEATURES_PROVIDED will abort compilation (as a required feature is not available).
Everything in FEATURES_REQUIRED and ```FEATURES_OPTIONAL & FEATURES_PROVIDED``` will be added to usemodule.

> So what is the proper way to include peripherals, 

So in your application, just specify all required features in FEATURES_REQUIRED, and they'll be used.

> and why the discrepancy with periph_uart?

uart_stdio, which is selected by default, adds periph_uart to FEATURES_REQUIRED.

photonthunder, 2018-02-06:
Thanks for the information, so is there some documentation that explains this?  I didn't even realize there was three options (REQUIRED, OPTIONAL, and PROVIDED).

closed 2019-09-10 by stale[bot]