RIOT-OS/RIOT #7972: interests of periph_init() function ?

dylad, 2017-11-09:
I would like to understand the use of  periph_init function within cpu.c file in order to initialize all available SPI.
The call of periph_init.c is handled by the drivers/periph_common/init.c

Why initialize all available SPI here ? Shoudn't we let the user & his app doing it ?

haukepetersen, 2017-11-09:
1 -> please refer to the doxygen of the `spi_init` function in `spi.h`

In short: as SPI is a shared bus, it needs to be initialized independent from a device/bus-users, and the simplest way we found is doing this during device startup. As the device is put to sleep after initialization, this should have minor/no penalties on energy usage (taken the periph drivers are implemented correctly), and this is much cleaner than having each bus-user call `spi_init` with some construct like `if (is_initialized_alread) then return ...`.

dylad, 2017-11-09:
Thanks @haukepetersen for the explaination.
But then SPI is re-initialize when we use a driver like `sx127x_driver`. Maybe there is some optimizations to do here ? Or am I missing something ?

gebart, 2017-11-09:
@dylad if a driver calls spi_init from its device init, then it is broken and should be fixed. The device drivers should however call the spi_init_cs function to set up the CS pin state, but the bus should be left alone.
AFAICS the sx127x driver only calls spi_init_cs, but never spi_init.

dylad, 2017-11-09:
aaaaaalright then I understand. I get it wrong since the beginning.
Thanks @gebart  :)
I'm closing it.

closed 2017-11-09 by dylad