RIOT-OS/RIOT #7978: What is the state of the x86 port?

haukepetersen, 2017-11-09:
Every once in a while I stumble over the `x86` port and wonder, what the state and our future ambitions with this are. At least on my local machin, the `x86-multiboot-common` does not even build ("cpu_conf.h not found), and from a code/structure perspective this port is not really the best state. Also it seems that there is not really any activity around this port for the last 2 years (hope I did not overlook anything).

So just asking: does it makes sense to (at least for now) remove the port?

Kijewski, 2017-11-14:
>  What is the state of the x86 port?

Abandoned and nonworking. I tried to remove it from the tree but that caused other problems so I gave it up pretty soon.

haukepetersen, 2017-11-15:
Thanks for the heads up! So would you be ok if I'll give it another try for removing it?

haukepetersen, 2017-11-15:
For the sake of documentation, it also seems like all x86 related issues (#1504, #3630, #2300) are not really active...

emmanuelsearch, 2017-11-15:
So, shall we remove aim to remove the support officially, since it is broken and no one champions making it work again?

haukepetersen, 2017-11-15:
I'd say so. As the existing code is clearly unused and broken for a long time and there was no need for fixing it, it seems that demand for this feature can't be too great... Also as nobody feels like putting in time at the moment, removing the port seems like the valid option. And as the removal is well documented (-> `LOSTANDFOUND.md`), we can always fix and pull the port back in.

PR #8042 (2017-11-16): boards/cpu: remove support for x86
fixes #7978

Dry run trying to remove the `x86` port from RIOT.

Note: a separate PR adding the removed modules to the `LOSTANDFOUND.md` will be created once/if we have a consensus on this PR.
closed 2017-11-16 by kaspar030