RIOT-OS/RIOT #8000: stm32/pwm: timer multiplier not taken into account

george-hopkins, 2017-11-10:
On some platforms (e.g. STM32F103xE), the clock for the timers is multiplied by 2 whenever the APBx prescaler is not 1. As far as I can see, the multiplication is currently not taken into account by the PWM driver (the APBx frequency is used directly). This results in a wrong calculcation of the timer prescaler (PSCx) and a PWM frequency which is too high.

#### Example (frequency set to 500 Hz)
[image]

#### Clock Tree for STM32F103xE
[image]

haukepetersen, 2017-11-10:
yapp, this is indeed an issue - I think we had the same for the `timer` implementation. Would you be so kind and provide a fix for the `pwm` driver? Thx.

haukepetersen, 2017-11-10:
I think the solution here would be to generalize the `apbmul[]` part from the `timer.c` driver and use it for both the `pwm` and the `timer` driver...

miri64, 2017-11-13:
With #8001 merged, is this issue fixed?

aabadie, 2017-11-13:
Yes, I'm closing it.

closed 2017-11-13 by aabadie