RIOT-OS/RIOT #8003: Ping fails from border router with slip

kb2ma, 2017-11-10:
I am unable to ping from a samr21-xpro border router to my Ubuntu 16.04 workstation. I use slip and the tunslip tool for serial communication. See the gcoap [Makefile.slip](https://github.com/RIOT-OS/RIOT/blob/master/examples/gcoap/Makefile.slip) for setup. I know I need to move on to ethos, but I would expect slip to be working. Have I missed something?

You can see [README-slip.md](https://github.com/RIOT-OS/RIOT/blob/master/examples/gcoap/README-slip.md) for the procedure. I'm leaving out RPL for this test, so I just do the steps below. The board is bbbb::2, and the tun interface is bbbb::1.

```
$ make term BOARD="samr21-xpro" PORT="/dev/ttyACM0"
>ifconfig 8 add unicast bbbb::2/64
>ncache add 8 bbbb::1
>ping6 bbbb::1

```
I did a little experimenting to find a good build.

| Branch (commit)  | Result |
| --- | --- |
| master on 10 Nov (d2ccbc8) | fail |
| 2017.10-branch (317e871)  | fail |
| master on 1 Oct (67244f2) | OK |

bergzand, 2017-11-28:
@kb2ma Is this still an issue now that #7925 is merged?

kb2ma, 2017-11-29:
Yes, this still times out as of commit 156240c from today. The ncache command is gone from the shell, so I use the following instead:

`nib neigh add 8 bbbb::1`

bergzand, 2017-11-29:
Thanks for testing, I'm going to try to reproduce the issue.

bergzand, 2017-11-30:
@kb2ma I can reproduce the issue on a nucleo-f446. I don't have a solution yet.

From what I can see with some debugging is that my Linux workstation replies to the ping requests, but on RIOT the replies are lost at either the slipdev driver or at the layer above.

kb2ma, 2017-11-30:
Thanks for verifying, Koen.

miri64, 2017-12-13:
@kb2ma @bergzand was you host configured as a router or did you set-up a default router to the host?

miri64, 2017-12-13:
Sorry, should have read the readme first ...

kb2ma, 2017-12-13:
I'm just trying to ping between the BR mote and my workstation, which is not set up as a router. Can you recreate the problem too, @miri64? It would be helpful to get this working again to move some of the gcoap PRs forward.

miri64, 2017-12-13:
> Can you recreate the problem too, @miri64? 

I can recreate it as well. Interestingly enough, I'm getting a reply, when I dump the packet on send...

> It would be helpful to get this working again to move some of the gcoap PRs forward.

Why aren't you using the usual border router `examples/gnrc_border_router` btw?

miri64, 2017-12-13:
For some reason the `RX` callback of the UART is not called or skipped :-/

kb2ma, 2017-12-13:
The gcoap version of the border router makefile and README are based on `examples/gnrc_border_router` when that example was based on slip. It's easier/valuable to test with a single mote instead of setting up two motes, where the border router is intermediary.

Glad to hear you can recreate the issue.

miri64, 2017-12-13:
It's weird... When I apply the following patch

```diff
diff --git a/drivers/slipdev/slipdev.c b/drivers/slipdev/slipdev.c
index 80ee16e..5992435 100644
--- a/drivers/slipdev/slipdev.c
+++ b/drivers/slipdev/slipdev.c
@@ -118,6 +118,7 @@ static int _init(netdev_t *netdev)
 
 static inline void _write_byte(slipdev_t *dev, uint8_t byte)
 {
+    printf("%c ", byte);
     uart_write(dev->config.uart, &byte, 1);
 }
 
@@ -149,6 +150,7 @@ static int _send(netdev_t *netdev, const struct iovec *vector, unsigned count)
         }
     }
     _write_byte(dev, SLIP_END);
+    puts("");
     return bytes;
 }
```

It works, even after I revert that patch it works and I can't reproduce it anymore... Can somebody else try that? I think I start to loose my sanity ^^

miri64, 2017-12-13:
Ok, when I restart tunslip during a ping, I get into the described behavior again oO...

bergzand, 2017-12-13:
I'm quite certain that I had the SLIP code working at least once when testing, but I also know that it is not the most stable protocol/code. I'm willing to investigate as soon as I have time to play with it (probably around Christmas).

kb2ma, 2017-12-13:
I plan to confirm later today. I agree that SLIP in this context has a bad reputation. It might be valuable to understand what triggered the instability we see now though.

miri64, 2017-12-13:
Confirmed, that it is not the `RX` callback. I'm able to "fix" it with the following patch as well (`_recv` is triggered by the UART RX callback):

```diff
diff --git a/sys/net/gnrc/netif/gnrc_netif_raw.c b/sys/net/gnrc/netif/gnrc_netif_raw.c
index 7abb6cb..53a81f3 100644
--- a/sys/net/gnrc/netif/gnrc_netif_raw.c
+++ b/sys/net/gnrc/netif/gnrc_netif_raw.c
@@ -53,6 +53,7 @@ static gnrc_pktsnip_t *_recv(gnrc_netif_t *netif)
     int bytes_expected = dev->driver->recv(dev, NULL, 0, NULL);
     gnrc_pktsnip_t *pkt = NULL;
 
+    puts("r");
     if (bytes_expected > 0) {
         int nread;
```

kb2ma, 2017-12-14:
@miri64, the patch above to gnrc_netif_raw.c works for me. However, the fix in #8258 does not. In both cases, I follow the steps at the top of this issue. I am testing on a samr21 Xplained Pro board.

Unfortunately, I don't have enough experience at this level to offer advice. However, I really appreciate the workaround above to allow testing gcoap. I am happy to try out any other fixes.

miri64, 2017-12-14:
@kb2ma can you test #8268?

kb2ma, 2017-12-14:
No problem, will try #8268 later today.

PR #8268 (2018-01-09): slipdev: simplify and solidify byte-unstuffing
### Contribution description

This simplifies and solidifies the reversal of SLIP's byte-stuffing
(aka byte-unstuffing ;-)) by

1. Using `tsrb` instead of `ringbuffer`: there are two actors. The ISR
   and the device's event handler.
2. Moving the byte-unstuffing from the UART RX-handler (i.e. the ISR)
   to the device's receive function (potentially not the ISR)
3. Removing the `pktfifo` member. The current number of bytes in the
   ringbuffer is returned for `recv(data = NULL, len = 0)`. If that is
   more than the packet contains (due to the by [... cut]
closed 2018-01-09 by bergzand