RIOT-OS/RIOT #8019: Module 'cpu' name conflict for `mips_pic32mx` and `mips_pic32mz`

cladmi, 2017-11-13:
I try to removes cases where multiple built modules have the same name.
It should not exist, and this would allow more reliable build: https://github.com/RIOT-OS/RIOT/pull/7952

Right now I have issues with `mips` boards.

`mips_pic32mx` and `mips_pic32mz` implement the `cpu` module but also depend on `mips32r2_common` which also implements it (and then also from its `periph` directory).
I could say, just rename `mips32r2_common` from `cpu` to `mips32r2_common` but it is also, despite its `_common` name, a standalone `cpu.

In `mips_pic32mz/mx` I see the following include
```
mips_pic32mx/eic_pic32mx.c:#include "../mips32r2_common/include/eic_irq.h"
```
which makes me also think that we are doing something wrong.

So my idea would be to add another cpu directory, `mips32r2_malta_something_other_name` and separate what is specifically a cpu, and what is in common.

From what I understood, `mips32r2_common/*.c`, `mips32r2_common/include/eic_irq.h` and `mips32r2_common/periph` and `newlib_syscalls_mips_uhi` should be in common. Some part of Makefile.include could be too.

And then, the "new" cpu would basically only have the `*cpu*.h` includes and its specific `Makefile.include`.

I have a question though on `Makefile.features`, right now `mipc_pic32mz` and `mips_pic32mx` do not include `mips32r2_common/Makefile.features` but still use its periph, so I think its a mistake.

I will propose a solution implementing what I say.

@neiljay I would be interested in help for naming convention and also in a short description for each cpu directory, I do not really understand the difference between all the cpus and why both mips32r2_common and mips_pic32_common are needed.

cladmi, 2017-11-13:
I do not understand why `mips_pic32_common/Makefile.include` adds

`INCLUDES += -I$(RIOTCPU)/mips32r2_common/include`.

Its unused right now because the only include that could be important is `eic_irq.h` which is done differently.
But that makes me wonder about what are the dependencies between the different boards common implementation.

cladmi, 2017-11-13:
I have a solution put into https://github.com/RIOT-OS/RIOT/pull/7952
I want to check some intermediate commits and will do a proper for it PR tomorrow.

neiljay, 2017-11-14:
@cladmi,

Hi I'll try to review this today, but things are crazy at work at the moment.

In the mean time can we name the new cpu 'mips32r2_generic' (instead of mips32r2_malta).

Cheers,

Neil

cladmi, 2017-11-14:
I will rename it, I just picked malta as its used in malta board.

cladmi, 2017-11-14:
Something I do not understand, is that right now, `mips_pic32mx/mz` use two common mips boards, `mips32r2_generic` and `mips_pic32_common`. Is the dependency to `mips32r2_generic` already a dependency from `mips_pic32_common` or just the `pic32mx/mz`, and why? :)

cladmi, 2017-11-24:
@neiljay I am still interested on your expertise, can the dependency to `mips32r2_common` be moved to `mips_pic32_common` ? or is it just specific to the mips_pic32mx/mz.

neiljay, 2017-11-24:
yes if the buildsystem allows you, all pic32 devices will depend on pic32_common and on mips32r2_common (all pic32 devices use a mips32r2 compatible core).

cladmi, 2017-11-24:
I can propose a PR for this, basically `pic32_mx/Makefile.include` will include `pic32_common/Makefile.include` which includes `mips32r2_common/Makefile.include`.

closed 2017-11-28 by neiljay