RIOT-OS/RIOT #8024: boards/cpu: nucleo-l152 (stm32l1) broken on master

haukepetersen, 2017-11-13:
While testing #7687 we discovered, that the `nucleo-l152` (`stm32l1`) is broken since quite a while now. Bisecting lead to the changes in #7385 (specifically to dd49f225321d3929f41c6108749c04eab0eb9571).

The change from `__irq_enable()` to `irq_restore(state)` breaks the L1 for some reason, can't really tell why... 
When doing `s/irq_restore(state);/__enable_irq();/`, everything is working (again). The difference seems to be `/__ASM volatile ("MSR primask, %0" : : "r" (priMask) : "memory");`
vs
`__ASM volatile ("cpsie i" : : : "memory");`

 Any ideas?

smlng, 2017-11-13:
I tried my fix for the TI CC2538 and CC2650 (see #7752) where I had a similar issue but that didn't work 😞

vincent-d, 2017-11-13:
I don't have any stm32l1 to test, but I don't see any reason `__enable_irq` and `__set_PRIMASK(0)` would have different behaviors.
Both are supposed to clear the `primask` bit (see stm32 programming manual: http://www.st.com/content/ccc/resource/technical/document/programming_manual/5b/ca/8d/83/56/7f/40/08/CD00228163.pdf/files/CD00228163.pdf/jcr:content/translations/en.CD00228163.pdf).

Could you debug a bit more the hard fault?

gebart, 2017-11-13:
check the status of the primask bit when you are in a normal thread (put a breakpoint in main or sth), and also around the disable irq and WFI stuff in pm.c

smlng, 2017-11-14:
compiling with `CFLAGS += -DDEVELHELP` helps, I don't get any faults then - maybe some misled compiler optimisation or thread stack sizes are _fixed_ that way.

paula75, 2017-12-13:
I have a nucleo-l152 and a nz32-sc151 both have the same cpu(stm32l1), i run the test examples: periph_adc, xtimer_msg and periph_rtc, this is what i get:

- Master branch:
Without changing anything all 3 test works on nz, but none in the nucleo. When i change irq_disable and irq_restore to disable_irq and enable_irq the periph_rtc works, but periph_adc and xtimer_msg are broke. 
- 2017.10 branch: 
Again, all 3 test works on nz, but none in the nucleo. When i change irq_disable and irq_restore to disable_irq and enable_irq all 3 test are working again.

From this i get that there is another bug from cpu.h in master branch and i'm curious about running the test in nz32 that has the same cpu

kYc0o, 2017-12-25:
I see that the problem described here is very possibly the original cause of the broken board. Just for the record, I'd like to point what I figured out on #8240, on which disabling periph/rtc solves partially the problem. I say partially because #7712 doesn't work by disabling rtc, thus I think this issue describes more precisely the problem. However, I'd like to know how these situations are related (rtc vs irq).

kYc0o, 2018-01-07:
After looking more closely to the bug, I found that  `__WFI()` should be called differently. I don't know why for other architectures it still works and only for the L1 it's a problem, but if I comment out the call to `__WFI()` and leave the current interrupt enable/disable instructions everything works correctly.

For more information about the usage of `__WFI()` I found [this](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.ddi0337e/ch05s09s02.html) and in [this](https://web.eecs.umich.edu/~prabal/teaching/eecs373-f10/readings/ARMv7-M_ARM.pdf) manual on the page 469 or B1-51.

Normally, `__WFI()` should be called in a loop in thread mode.

gebart, 2018-01-08:
@kYc0o WFI is being called in a loop when there is nothing else going on in the system. The idle thread only contains a loop which calls the pm functions which call WFI on cortexm.

kYc0o, 2018-01-08:
Ok I see @gebart. However I still think there's maybe a misuse of the interrupt enable/disable since we've a regression to #6484, which states more or less the same problem than this issue.

kYc0o, 2018-01-11:
Well after a quite long research and efforts to understand the sleep modes on Cortex-M platforms, I realised that the wake-up requirements are kind of different depending on the CPU. If someone can clarify the decisions made to make it that generic, I'd be glad to read about it. Also, maybe all of them mean more or less the same, and so I'm just unable to understand exactly what they mean so if the implementors of the PM can guide me I'll be very thankful.

For Cortex-M0 and M0+:

```
*Wakeup from WFI or sleep-on-exit*
Normally, the processor wakes up only when it detects an exception with sufficient
priority to cause exception entry.

Some embedded systems might have to execute system restore tasks after the
processor wakes up, and before it executes an interrupt handler. To achieve this set
the PRIMASK bit to 1. If an interrupt arrives that is enabled and has a higher priority
than current exception priority, the processor wakes up but does not execute the
interrupt handler until the processor sets PRIMASK to zero. For more information
about PRIMASK, see Exception mask register.
```

for Cortex-M3 (our case here so far) and M4:

```
*Wakeup from WFI or sleep-on-exit*
Normally, the processor wakes up only when it detects an exception with sufficient
priority to cause exception entry. 

Some embedded systems might have to execute system restore tasks after the
processor wakes up, and before it executes an interrupt handler. To achieve this set
the PRIMASK bit to 1 and the FAULTMASK bit to 0. If an interrupt arrives that is
enabled and has a higher priority than current exception priority, the processor
wakes up but does not execute the interrupt handler until the processor sets PRIMASK
to zero. For more information about PRIMASK and FAULTMASK see Exception mask
registers.
```

However, for some reason only the L1 is being affected. And still 2017.10 works with this.

Further tests indicate that in current master even changing the irq functions the board still doesn't work.

Edit: (I didn't send the message because I didn't hit the button)
Actually the problem is very simple... Revision C-01 doesn't have external low frequency crystal, C-03 has it and works properly on master.

kaspar030, 2018-01-12:
> Well after a quite long research and efforts to understand the sleep modes on Cortex-M platforms, I realised that the wake-up requirements are kind of different depending on the CPU. If someone can [...]

What do you mean? PRIMASK is set through irq_enable()/irq_disable(). Both platforms do on WFI sleep until an interrupt occurs. If ISRs are enabled, they jump right in. If not, they continue after WFI. Are you referring to FAULTMASK?

kYc0o, 2018-01-12:
> Are you referring to FAULTMASK?

Yes, is the difference I want to highlight.

kaspar030, 2018-01-12:
> > Are you referring to FAULTMASK?
> Yes, is the difference I want to highlight.

AFAIK we don't set FAULTMASK anywhere, and apart from that, the cortexes should behave the same...

kYc0o, 2018-01-15:
I modified the current code to set FAULTMASK to 0 when awaking and everything works again. I'll make a PR with the fix.

kYc0o, 2018-02-05:
After a debugging session with @cladmi we found that it was the variable `state` which is messing everything up on stm32l1. Why only on that CPU? I'm investigating, but what we know now is that the variable `state` is being stored "temporary" in r0 (maybe because of compiler optimisations), which after sleep apparently is lost. The solution is to force the variable to be stored in the stack by declaring it `volatile`. We'll (@cladmi ?) submit a PR with the fix, then we need to test if it doesn't break anything on other platforms.

PR #8545 (2018-03-01): boards/nucleo-l152: configure LSI by default

### Contribution description
Some nucleo-l152 boards, namely revision C-01, don't have a LSE to rely on for low speed clocking, which is mostly used for RTC.

This PR makes it optional, and define LSI by default.

### Issues/PRs references
Fixes #8024 and fixes #8240, depends on ~#8518~ to fix everything.
closed 2018-03-01 by aabadie