RIOT-OS/RIOT #8046: netif2/raw: Don't enable router advertisements for slipdev interfaces

bergzand, 2017-11-15:
Router advertisements are enabled by default on the slipdev interface. to me, this doesn't really make any sense as the main use of the interface is to provide connectivity via a controlling host. Is there a case where router advertisements over SLIP are useful?

Can't say it disrupts the connection, but it looks like unnecessary traffic, (thus minor and I wouldn't consider this blocking for the whole netif2 integration).

miri64, 2017-11-15:
Why shouldn't the border router advertise itself as a router in your opinion? Or am I missing something?

miri64, 2017-11-15:
To me, if you connect your board via SLIP (or Ethos for that matter) to a "controlling host" you have basically two hops from the WPAN to the rest of the world: The border router and the host system. I state this because maybe there is some misunderstanding that the board is just acting as a network interface for the host system (I deduce this from your wording "controlling host").

bergzand, 2017-11-15:
Maybe I'm associating router advertisements too much with SLAAC here. As I see it, with SLIP the interfaces on both sides of the serial link are statically configured, both addresses and routes, and no advertisement of router functionality is necessary.

> To me, if you connect your board via SLIP (or Ethos for that matter) to a "controlling host" you have basically two hops from the WPAN to the rest of the world: The border router and the host [...]

"controlling host" is a bad choice of words here. I'm aware of the two hops and the network between the host system and the border router. In my experience this network between the host system and the border router is statically configured and explicitly advertising router functionality is not useful. 

Now if I'm violating an RFC with this issue or if you think it is better to leave them enabled, I'd can agree with that, it's just that it always appeared to me as needless traffic.

miri64, 2017-11-15:
RAs are not just used for SLAAC. Even the PIO -  the option that carries the relevant address information - isn't just for SLAAC (it also updates the node's prefix list which is used to help a node to determine addresses that are on-link). The main function of the RA is what its name suggests: to advertise a router to its neighbors. I wouldn't say it necessarily breaks any RFC (after all the SLIP link is a very restricted link and what you do on your own link is in the end up to you). I just think it's weird to not advertise the router.

Since this is a question of configuration (one can always Unser the RTR_ADV flag on a router), I would say we close this issue for now and deactivate that flag per default, if the RAs provenzalisches to be a problem (which they shouldn't, as they are supposed to be send in larger intervals than high load use data).

bergzand, 2017-11-15:
I can agree with that. Thanks for the explanation, learned something today.

miri64, 2017-11-15:
(written from a phone so excuse some of the weirder spelling errors caused by auto-completion. Can't fix them right now.)

closed 2017-11-15 by miri64