RIOT-OS/RIOT #8048: Riot on raspberry pi using it's pinout

florian-jean, 2017-11-16:
Hi all,

I was trying to implement my MRF24j40 on my raspberry without the need to slip the radio nor making all the routing set up myself as it is found [here](https://github.com/RIOT-OS/RIOT/wiki/How-to-install-6LoWPAN-Linux-Kernel-on-Raspberry-Pi)

So I stumbled upon [this fork](https://gitlab.ibr.cs.tu-bs.de/hartung/RIOT/tree/yschroed/boards/raspberrypi). Unfortunatly it didn't compile on my raspberry and was outdated by RIOT rework on cpus file agencement ...

I used some part of the files of yschroed and some part of the files of native and rewote some spi/gpio handler into the current version of riot of my pc

It compiles with WERROR=0 but it is not clean and does not follow completely RIOT's way of coding.
So, if somebody having a knowledge advanced enough on RIOT (not me unfortunatly) can spend an hour making it clean and merging the code, that would be great for the community, especially as raspberry is widely used, I find that it's a shame having to use a slip instead of the gpio

The board and cpu files are attached
[riot.zip](https://github.com/RIOT-OS/RIOT/files/1477836/riot.zip)

miri64, 2017-11-16:
Hi,

RIOT is in general not targeting platforms that can fit operating systems like Linux (or that are not microcontrollers). In the end I think it is easier to just go for the Linux-based border router [setup found in the *updated* version of the wiki page you linked](https://github.com/RIOT-Makers/wpan-raspbian/wiki) instead of trying to fit RIOT on a platform it really isn't meant for. Hope you understand.

Cheers,
Martine

kaspar030, 2017-11-16:
> I was trying to implement my MRF24j40 on my raspberry without the need to slip the radio nor making all the routing set up myself as it is found here

Do I get it right that the linked code still runs a riot instance on linux, but adds support to use the RasPi's gpio directly?

miri64, 2017-11-16:
> Do I get it right that the linked code still runs a riot instance on linux, but adds support to use the RasPi's gpio directly?

Oh, I didn't look that deeply into the code, but that would indeed be interesting (but I wouldn't call this board `raspberrypi` then, but put it in `native`).

bergzand, 2017-11-16:
@florian-jean I think that I've once build in RIOT what you describe here.
I used sysfs GPIO and spidev code to be able to control an mrf24j40 from Linux userspace. Then add a GPIO implementation (#7530) and a Linux spidev spi handler implement for the RIOT native platform. Add the correct params.h and you've got an RIOT native instance with a working radio attached. I've never really got it stable, I suspect threading issues.

bergzand, 2017-11-16:
Stable as in, remain working for multiple days, it panic/segfaulted/assertion failure after a day or so.

florian-jean, 2017-11-16:
@miri64 Yes I understand that using the appropriate tools on linux would be nice but until it's not implemented natively in the kernel, it would be nice to have that option and this will ease the set up of RIOT for raspberry pi users.

@kaspar030 Yes, it does. Just but the board and cpu folder of native_pi into the respective folder of riot and use the board native_pi. Beware that the pins numeration is quite different and can [here](https://pinout.xyz/pinout/wiringpi#). You can change the pin numeration (GPIO or hardware numeration) by modifying the set up function [eg this](https://projects.drogon.net/raspberry-pi/wiringpi/functions/) but this may have implications about running with/without root privilege and accessing CPU register, so security issue if set up handled properly

@bergzand I dunno about the stability of the code over a long period of time, but it uses wiringPI to access GPIO and SPI, so it may be worth a look. 

@miri64 As the library used (wiringPI) is only on raspberry pi, the library needs conditional including and the spi and gpio handlers too

bergzand, 2017-11-16:
> @miri64 As the library used (wiringPI) is only on raspberry pi, the library needs conditional including and the spi and gpio handlers too

This is the reason I opted for using sysfs gpio and Linux spidev. This way it should also work on the beaglebone or any other Linux board with GPIO and spi support. The main problem I had is that Linux spidev doesn't support extending the spi cs active between transfers, but that's solved by doing software spi cs. I'll take a look at the code, might be interesting to see if things are done better there, although I don't like the idea of depending on wiringpi for GPIO access.

@florian-jean If you (and the others participating in this issue) think that it is useful to have GPIO and SPI support for a native platform, I can continue working on this a bit more and see if it is good enough to get it merged.

miri64, 2017-11-16:
> @miri64 Yes I understand that using the appropriate tools on linux would be nice but until it's not implemented natively in the kernel, it would be nice to have that option and this will ease the set up of RIOT for raspberry pi users.

As I said: I might have misunderstood your request, as you were asking for putting RIOT on Raspberry Pi. As the fork you reference just seem to do `native` + some rasppi stuff, it is something completely different (running RIOT as a process on Linux) as I understood it (to be able to run RIOT on Raspberry Pi without any Linux under it, as we usually do it for other hardware).

florian-jean, 2017-11-16:
@bergzand I haven't through much about portability to other linux platform indeed.

I think gpio and spi support would be nice in order to make border router (which is memory intensive) run on actual computer instead of a MCU that slip packets

@miri64 yeah we misunderstood each other :-)

roberthartung, 2017-12-05:
We did something similar once - https://gitlab.ibr.cs.tu-bs.de/hartung/RIOT/tree/yschroed/boards/raspberrypi this might help you

miri64, 2017-12-05:
@roberthartung that repository was linked in the original post ;-)

roberthartung, 2017-12-05:
@miri64 Sorry :-D

roberthartung, 2017-12-05:
So the fork was created by a colleague and not maintained. We might jump back on the train though and it might be really cool to have RIOT be running as an application on a raspberry pi. +1

closed 2019-09-10 by stale[bot]