RIOT-OS/RIOT #8052: mips: several issues

kaspar030, 2017-11-16:
Apart from the issues in #6588, there are some more:

- [ ] the mips CPU makefiles have quite a lot of code duplication
- [ ] the mips flash targets are empty. They should fail and explain why ("No flash tool available. See https://..." ...).
- [ ] the mips board wiki-pages lack information on how to flash.

neiljay, 2017-11-16:
@kaspar030 (and everyone else),

I will  attempt to address these issue, but it may take a couple of weeks, things are very busy at the moment, In case you had not heard:

https://www.electronicsweekly.com/news/business/imagination-press-ahead-mips-sale-us-group-tallwood-2017-09/

Neil

cladmi, 2017-11-30:
This PR fixed some of the code duplication in the makefiles: https://github.com/RIOT-OS/RIOT/pull/8032
Some more cleanup can be done by making `mips_pic32_common` depend on `mips32r2_common` which would depend on `mips.inc.mk`.

gebart, 2017-12-08:
Some more issues popped up:
 - [ ] The MIPS CPUs never call the common periph_init during boot as far as I can tell.
 - [ ] The MIPS CPUs do not have a cpu_init function which is called by the board_init to initialize things in the CPU, (this function in the other CPUs is sometimes very simple with only a call to periph_init)

neiljay, 2018-06-27:
item 2 now resolved in #9259. I think this can be closed ?

cladmi, 2018-06-28:
It is only flashing `pic32-wifire` and not `pic32-clicker`, also it is currently not merged :)
But we are getting closer to closing this.

kaspar030, 2019-09-10:
This is still being worked on.

PR #13125 (2020-01-15): boards: pic32-clicker: Use pic32prog as default programmer
### Contribution description

This PR adds a default programmer for the pic32-clicker board such that you can use `make flash` to program your board.

### Testing procedure

I flashed `hello-world` example on a pic32-clicker and I verified that I received the right message on the serial port.

### Issues/PRs references

This should close #8052.
closed 2020-01-15 by aabadie