RIOT-OS/RIOT #8059: examples/* in general - missing/duplicated/confusing - hierarchy - discussion

temmihoo, 2017-11-16:
The examples/ folder has some shortcomings. As an effort to make better collection of more explanative examples, I started this issue by request instead of just chatting shortly on IRC.

Documentation
===========
Some examples are lacking in the documentation level they provide. This in my view should be remedied by both commenting the code and having a README.md in all of the examples. Naturally the entity names in code should be well chosen but as the aim is to be explanative, good comments and pointers to reading material would not be wrong.

Misleading
========
This is more complex. For example the examples/saul does not really help anybody in learning how saul interfaces are programmed against. It just briefly mentions that you can potentially something something with something called saul in the shell. I think that this example should show how peripherals are interfaced with saul - along with documentation pointers and mentioning that the shell commands also exist. As of now this is completely useless for anybody not knowing a lot of RIOT.

Hierarchy
=======
This has been discussed before so sorry about opening a can of worms. We are having a lot of networking examples and very little non-networking examples. I think that as our aim should be to have a comprehensive set of examples, the sheer number of them necessitates categorising them and putting them in some sort of hierarchy. Now that the number is small we should discuss on how to start this categorisation before it is absolutely necessary to just shove them somewhere.

My view is that all APIs should have a simple yet comprehensive example of how it is used. This is not very far from an unit test so it is fully possible that some of the examples might share code with tests. If the example starts getting very complex it is also telling that either the API design is skewed wrong or that the thing really is complex and warrants a complex example. I think that all API examples should be grouped together using the same hierarchy that the API code is.

Then there are the examples that are more like use case examples than API examples. These need not be as comprehensive in that they do not need to showcase all of any single API or feature. This category would be good learning material for somebody to have an embedded systems class in a Hacklab or school or something.

So about hierarchy discussion I am proposing:
examples/API
  mirroring the existing hierarchy
examples/Networking
  most of current examples fall in here
  subcategories by protocol could well symlink to the API examples
examples/Scheduler
  several examples at different complexities to showcase how the scheduler works
examples/Callbacks
  ISR callbacks of different types
examples/Power Management
  Power management personified by things that fall partially asleep at times
examples/Shell
  These are showing how to extend shell commands and how to interface with RIOT shell
  from your host system via usbserial or the CoAP shell
examples/Tutorial
  Here we have stuff from single LED blinkers to CoAP controlled PID heaters and the like.
  This should look like classroom material for different level of teaching for people who might not
  understand anything at the assignment start and will _potentially_ learn various things or possibly
  just do whatever experiment was at hand with zero learning - just like in classroom.

gebart, 2017-11-17:
I agree that it would be nice to have some more example code, like tutorial projects etc. However, I don't agree that the situation is as bad as you describe. The API usage examples are usually found under /tests/, where there is a small test application for every available device driver and a lot of system modules as well.
Could it be a useful addition to the Doxygen docs to link the test application code into each page describing a module, for a practical usage example?
I mean code such as this, which is pretty much the smallest runnable example of using this particular device driver: https://github.com/RIOT-OS/RIOT/blob/master/tests/driver_ina220/main.c

What is your opinion of the tutorials and guides in the wiki? (https://github.com/RIOT-OS/RIOT/wiki#tutorials-and-howtos)
I have to admit that I have not read through most of them, but someone has taken the time to write quite long descriptions for different application use cases. If the problem is visibility, then I think we should improve the location and linking for the tutorials, and possibly integrate them into the repository.

aabadie, 2017-11-17:
> This would include blinking a LED, showcasing how the scheduler allows for sleeping until rtc isr callback is needed and so on.

+1

closed 2019-09-10 by stale[bot]

miri64, 2019-09-10:
There is somewhat of an overlap to https://github.com/RIOT-OS/RIOT/issues/5105.