RIOT-OS/RIOT #8085: Update solicited node multicast groups on IPv6 address addition/deletion

bergzand, 2017-11-19:
Currently, no extra solicited node multicast addresses are added when an extra IPv6 address that is not derived from the L2 address is added to an interface. This breaks NS/NA.

Steps to reproduce:
1. Start two instances of gnrc_networking
2. Add an extra IPv6 address to the interface of the first node (for example 2001:db8::1/64)
3. Add same prefix to the second node (for example 2001:db8::2/64).
4. Try to ping the address from the second node.

miri64, 2017-11-19:
Are you talking about 6Lo-based interfaces or Ethernet interface? On 6Lo-based interfaces [this is on purpose](https://tools.ietf.org/html/rfc6775#section-5.2). On Ethernet interfaces this would be a bug.

miri64, 2017-11-19:
>  On 6Lo-based interfaces this is on purpose.

That a global address wouldn't be reachable in that case, might be due to a different reason.

bergzand, 2017-11-19:
This was tested on native, so with ethernet interfaces.

miri64, 2017-11-19:
Ok. Will investigate tomorrow! :-)

miri64, 2017-11-20:
Should be fixed with #8089.

bergzand, 2017-11-20:
Is fixed with #8089

PR #8089 (2017-11-20): gnrc_netif: join/leave solicited nodes IPv6 address add/remove
Fixes #8085 by actually joining the solicited nodes group on adding a new IPv6 address to an interface. Likewise, the group is left, when no IPv6 address associated to that group is left on the interface.
closed 2017-11-20 by bergzand