RIOT-OS/RIOT #8088: Rapid spam of unicast RS messages when destination unavailable (crashed)

bergzand, 2017-11-20:
I'm debugging a situation where the border router crashed due to a failed assertion (more on this in a later).
When the border router crashed, the second node is rapidly sending out unicast RS messages to the crashed node. the source is alternated between the link local address and the GUA.

Hardware:
two nucleo-f446 boards with mrf24j40's attached. One configured as border router, the other as generic node.

I have yet to look this up in the RFCs, but to me this looks like an accidental DOS by "jamming" the spectrum.

bergzand, 2017-11-20:
I need to learn to read better, it's spamming neighbor solicitations, not router solicitations. 😞

bergzand, 2017-11-20:
Backtrace starting from `gnrc_ndp_nbr_sol_send`:
```
#0  gnrc_ndp_nbr_sol_send (tgt=tgt@entry=0x20001ad8 <_nodes+4>, netif=netif@entry=0x20001c88 <_netifs>, src=src@entry=0x20001c9a <_netifs+18>, 
    dst=dst@entry=0x20001ad8 <_nodes+4>, ext_opts=0x20001db4 <_pktbuf>) at /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ndp/gnrc_ndp.c:233
#1  0x080039f0 in _snd_ns (tgt=0x20001ad8 <_nodes+4>, netif=netif@entry=0x20001c88 <_netifs>, src=src@entry=0x20001c9a <_netifs+18>, dst=0x20001ad8 <_nodes+4>)
    at /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c:76
#2  0x08004a44 in _handle_rereg_address (addr=addr@entry=0x20001c9a <_netifs+18>) at /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/nib/_nib-6ln.c:218
#3  0x0800367a in gnrc_ipv6_nib_handle_timer_event (ctx=0x20001c9a <_netifs+18>, type=<optimized out>) at /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/nib/nib.c:349
#4  0x08002708 in _event_loop (args=<optimized out>) at /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/gnrc_ipv6.c:303
#5  0x0800032c in sched_switch (other_prio=6872) at /home/hydrazine/dev/RIOT/core/sched.c:203
#6  0x77777776 in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

miri64, 2017-11-20:
> I'm debugging a situation where the border router crashed due to a failed assertion (more on this in a later).

Is that failed assertion related to the one fixed in #8087 or completely different?

bergzand, 2017-11-20:
> Is that failed assertion related to the one fixed in #8087 or completely different?

Looks like that's the one in this case.

miri64, 2017-11-20:
The statement that the NS are coming from both the GUA and LL tells me that this happens after the initial bootstrap. Is this still reproducible with #8087 merged (e.g. by disconnecting the border router after the initial bootstrap and waiting for the next registration)?

miri64, 2017-11-20:
> Is this still reproducible with #8087 merged (e.g. by disconnecting the border router after the initial bootstrap and waiting for the next registration)?

Confirmed that it is ;-)

bergzand, 2017-11-20:
Good to know that I'm not the only one able to trigger the issue.

miri64, 2017-11-20:
Fixed in #8091

miri64, 2017-11-20:
Mhh #8091 only fixed it partially. As soon as the neighbor goes to UNREACHABLE, the node starts spamming again...

miri64, 2017-11-20:
@bergzand One question that came up for me while debugging:

> [...] the other as generic node.

Do you mean the default config of `gnrc_networking` (i.e. a 6LR) with that or a 6LN?

miri64, 2017-11-20:
> Mhh #8091 only fixed it partially. As soon as the neighbor goes to UNREACHABLE, the node starts spamming again...

Ah the backoff starts overflowing. This is a separate issue from #8091. As such: Will fix in a separate PR.

bergzand, 2017-11-20:
> Do you mean the default config of gnrc_networking (i.e. a 6LR) with that or a 6LN?

examples/gnrc_networking, adapted to work on a nucleo-f446 with an mrf24j40. So the `BOARD` changed, the module for the radio and some defines to get the radio working. That makes it a 6LR.

> Ah the backoff starts overflowing. This is a separate issue from #8091. As such: Will fix in a separate PR.

Check, I'm going to merge #8091 then.

miri64, 2017-11-20:
I'm checking, if it does for real, but #8093 should be the definite fix for that.

miri64, 2017-11-20:
Backoff is properly capped now, so I'm satisfied with the state of #8093.

bergzand, 2017-11-21:
Closed as #8093 is merged and so far I haven't seen the spamming behaviour anymore.

PR #8091 (2017-11-20): gnrc_ipv6_nib: fix retransmission timeout unit for address re-registration
Fixes #8088 by fixing the retransmission time from 60 milliseconds to 60 seconds ;-).
PR #8093 (2017-11-21): gnrc_ipv6_nib: various fixes regarding UNREACHABLE state
This PR fixes two drawbacks in how the UNREACHABLE state is currently handled in the NIB:

1. A router that is UNREACHABLE can't help you when you try to re-register your address to it, so don't do it ;-).
2. With the sent NS counted it came to the point were `1 << ns_sent` resulted in NULL, because `ns_sent` was larger that the width of the type. With some calculations, I realized that 17 is the highest number `ns_sent` can get, before the backoff is higher than the specified truncation of 1 min.

The last commit is unrelated, but I did [... cut]
closed 2017-11-21 by bergzand