RIOT-OS/RIOT #8090: shell_commands ifconfig displays garbage with "holey" ip addresses.

bergzand, 2017-11-20:
I've configured a native `examples/gnrc_networking` node to have at least 3 IPv6 addresses. When the second of three addresses is removed, the third (or the second valid) address is displayed incorrectly:
```
> ifconfig
ifconfig
Iface  6  HWaddr: 42:a2:0a:45:a2:ec
          MTU:1500  HL:64  RTR
          RTR_ADV
          Source address length: 6
          Link type: wired
          inet6 addr: fe80::40a2:aff:fe45:a2ec  scope: local
          inet6 addr: ::f407:8e8:cd07:87f:bf04:878  scope: global
          inet6 group: ff02::2
          inet6 group: ff02::1
          inet6 group: ff02::1:ff45:a2ec
          inet6 group: ff02::1a
          inet6 group: ff02::1:ff00:1

          Statistics for Layer 2
            RX packets 14  bytes 2370
            TX packets 5 (Multicast: 5)  bytes 536
            TX succeeded 5 errors 0
          Statistics for IPv6
            RX packets 14  bytes 2174
            TX packets 5 (Multicast: 5)  bytes 466
            TX succeeded 5 errors 0
```

Steps to reproduce:
1. modify `sys/include/net/gnrc/netif/conf.h` `GNRC_NETIF_IPV6_ADDRS_NUMOF` to at least 3.
2. start `examples/gnrc_networking` on native.
3. `> ifconfig 6 add abcd::1/64`
4. `> ifconfig 6 add abce::1/64`
5. `> ifconfig 6 del abcd::1`
6. verify that `ifconfig` doesn't display abce::1 correctly

Pinging the remaining address (abce::1) from a different node still works correctly.

miri64, 2017-11-20:
Fixed by #8092

bergzand, 2017-11-20:
And indeed fixed with #8092.

PR #8092 (2017-11-20): gnrc_netif: fix GNRC_NETAPI_GET of arrays
Fixes #8090. The target array currently moves with the loop instead of with the results.
closed 2017-11-20 by bergzand