RIOT-OS/RIOT #8094: gnrc_udp: sending to link local address is discarded by gnrc_ipv6

bergzand, 2017-11-20:
I've been testing around with the issue discovered by @zhuoshuguo in #7925.
Testing was done using two native instances (examples/gnrc_networking), one running the udp server application, the other is used for the udp sending.

What I discovered so far is that transmitting to the GUA of the server node works as expected. Transmission to the link local address doesn't work, gnrc_ipv6 discards the packet at line 665.

@zhuoshuguo Can you confirm that this matches your issue reported in #7925?

miri64, 2017-11-21:
After looking at it myself, I think I know what is happening: The old forwarding + address resolution engine tried to resolve (wrongly) a link-local address without any interface given, the NIB doesn't do that anymore. However, the UDP command isn't adapted for that yet. Will do.

zhuoshuguo, 2017-11-21:
Yes, `Transmission to the link local address doesn't work` is also what I have observed. ;-)

miri64, 2017-11-21:
Should be fixed in #8095.

bergzand, 2017-11-21:
I would prefer to have a note of this change ~in~ to correct behaviour somewhere (release notes or doxygen), I can imagine applications relying on the previous behaviour just like the examples.

miri64, 2017-11-21:
@bergzand well the release notes only will be updated at the release, so I guess the best point would be the Doxygen for now. Where would you expect such a note?

miri64, 2017-11-21:
(it should also be noted, that the previous behavior, though convenient, was a bug that could cause NS-flooding / isn't possible with the optimizations for link-local addresses proposed in RFC 6775. The only reason it never had any implications was that we did not had any device in use that had the optimizations for link-local addresses in place *and* more than one interface)

bergzand, 2017-11-21:
I think a small note on the [GNRC overview page](http://riot-os.org/api/group__net__gnrc.html) should be sufficient. I think that would be the first place where I would look if I knew about this netif/nib overhaul. (Speaking of that page, is it still correct after this PR is merged?)

> (it should also be noted, that the previous behavior, though convenient, was a bug that could cause NS-flooding / isn't possible with the optimizations for link-local addresses proposed in RFC 6775. [...]

I agree that the new behaviour is better/more correct, but it is a change in behaviour, so there are probably some users that are going to complain about this change.

miri64, 2017-11-21:
> I agree that the new behaviour is better/more correct, but it is a change in behaviour, so there are probably some users that are going to complain about this change.

Yes. There is nothing regarding forwarding or address resolution on that page.

miri64, 2017-11-21:
@bergzand see #8103

miri64, 2017-11-21:
Since #8095 fixes the symptom documented in this issue and #8103 is a mere amendment, I'm closing this issue as fixed.

PR #8095 (2017-11-21): examples: tests: adapt udp shell commands for new forwarding engine
The NIB's forwarding engine behaves more correctly than GNRC's old
forwarding engine in that link-local addresses require also an
interface to be provided to be meaningful. As with other commands this
is simplified for the special case that there only is one interface
that that interface is chosen.

Fixes #8094.
closed 2017-11-21 by miri64