RIOT-OS/RIOT #8096: boards: nucleo boards not having their marketing names

kaspar030, 2017-11-21:
Currently, our nucleo boards are only roughly named like STM calls them.

E.g., my nucleo-f401re is just called "nucleo-f401". "nucleo-f446re" and "nucleo-f446ze" are called "nucleo-f446" and "nucleo144-f446".

I find that confusing, and so does google (e.g., a search for "RIOT nucleo-f446ze" doesn't show many RIOT results. "RIOT nucleo-f446" only shows the nucleo-f446re.)

I propose renaming all nucleos to whatever STM calls them.

aabadie, 2017-11-21:
I'm with you @kaspar030. According to the documentation on [ST website](http://www.st.com/en/evaluation-tools/stm32-mcu-nucleo.html), this should be the way to go.

Since I introduced the actual naming scheme, I can give you the reason of this initial choice: but the initial naming scheme, I was facing name conflicts between some nucleo64 and nucleo144 boards (especially the f446 one, but this also applies to the f303). I wanted to have a name that better reflect the type of board (nucleo32/64/144).

bergzand, 2017-11-22:
I agree with renaming the boards to their documented names from STM. I've had this problem reversed where I tried to look up a supported board on the STM website, but could get any results due to our nucleo(32|144) naming.

Would it be possible to use the board alias from #7976 to keep the old names as an alias and not break compatibility with existing applications? So rename every board to their marketed name and add an alias for the old name.

aabadie, 2017-11-22:
> Would it be possible to use the board alias from #7976 to keep the old names as an alias and not break compatibility with existing applications? So rename every board to their marketed name and add an alias for the old name.

+1, very neat idea

aabadie, 2018-02-27:
For those interested: #8650, #8649 and #8651

aabadie, 2018-05-23:
All related PRs were merged, closing this one.

closed 2018-05-23 by aabadie