RIOT-OS/RIOT #8099: gnrc: sporadic assertion failure at _nib-arsm.c:331

bergzand, 2017-11-21:
Sometimes I'm able to trigger the assert at `sys/net/gnrc/network_layer/ipv6/nib/_nib-arsm.c:331` on my 6LR (gnrc_networking example) node. I can't reproduce it consistently yet, so more information follows when I've triggered it a few more times.

setup:
2 nucleo-f446, mrf24j40 radio
1 configured as gnrc_border_router example mrf24j40 settings
1 configured as gnrc_networking example with mrf24j40 settings

miri64, 2017-11-21:
Can you try to print `_nib_onl_get_if(nbr)` (and maybe the IPv6 address of the neighbor using `ipv6_addr_to_str()` before the assert?

miri64, 2017-11-21:
@bergzand were you able to reproduce this bug with my proposed debug output?

bergzand, 2017-11-21:
Nope, nothing yet, everything as stable as it could be. 
I'm going to update my test nodes to the latest branch HEAD and keep them running for a while.

I still have absolutely no clue how I triggered this assertion, so I'm going to hope that I can randomly trigger it again and get some output.

bergzand, 2017-11-23:
@miri64 I haven't encountered this assertion since I created this issue. Do you mind if I close this issue tomorrow morning if my nodes didn't trigger it again by then?

miri64, 2017-11-23:
Not at all. :-)

bergzand, 2017-11-24:
Could not reproduce. Might be accidentally fixed by a previous PR.

closed 2017-11-24 by bergzand