RIOT-OS/RIOT #8105: cc110x: NETOPT_ADDR_LEN

hahchenchen, 2017-11-22:
when i run the ccn-lite-relay via cc110x, came out an error:
`  error: unable to determine address length for if=<3>
Error registering at network interface!`

The error cames from: ccn-lite-riot.c
`res = gnrc_netapi_get(if_pid, `NETOPT_ADDR_LEN`, 0, &(i->addr_len), sizeof(i->addr_len));
    if (res < 0) {
        DEBUGMSG(ERROR, "error: unable to determine address length for if=<%u>\n", (unsigned) if_pid);
        return -ECANCELED;
    }`

the `NETOPT_ADDR_LEN` is not defined in cc110x-netdev.c.
So anyone can help me?

miri64, 2017-11-22:
> the NETOPT_ADDR_LEN is not defined in cc110x-netdev.c.
So anyone can help me?

I guess that is the problem ;-). Will provide a fix soonish.

PeterKietzmann, 2017-11-22:
#8111 solves the issue

closed 2017-11-22 by PeterKietzmann