RIOT-OS/RIOT #8106: PKG wakaama fail to apply patched

florian-jean, 2017-11-22:
Hello,

I want to start using the wakaama pakage.
I copied the gnrc_minimal exemple and added to the makefile 

> USEPKG += wakaama

However, when I try to compile it, I get 

> Applying: fixed warnings in er-coap-13.c
error: patch failed: core/er-coap-13/er-coap-13.c:1231
error: core/er-coap-13/er-coap-13.c: patch does not apply

Full log

>  sudo make clean all
make: Nothing to be done for 'clean'.
Building application "gnrc_minimal" for "native" with MCU "native".
rm -Rf /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama
mkdir -p /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama
/home/florian/Documents/prog/riot_clean2/dist/tools/git/git-cache clone "https://github.com/eclipse/wakaama.git" "69a32cfae39f66fe4eec4cc8d1cd48ced7ad447c" "/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama"
Cloning into '/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama'...
remote: Counting objects: 5942, done.
remote: Compressing objects: 100% (9/9), done.
remote: Total 5942 (delta 1), reused 6 (delta 1), pack-reused 5932
Receiving objects: 100% (5942/5942), 2.14 MiB | 108.00 KiB/s, done.
Resolving deltas: 100% (4201/4201), done.
Checking connectivity... done.
HEAD is now at 69a32cf... Merge pull request #57 from eclipse/fix_observe
touch /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama/.git-downloaded
git -C /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama checkout -f 69a32cfae39f66fe4eec4cc8d1cd48ced7ad447c
HEAD is now at 69a32cf... Merge pull request #57 from eclipse/fix_observe
git -C /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama am --ignore-whitespace "/home/florian/Documents/prog/riot_clean2/pkg/wakaama"/patches/*.patch
Applying: changing header location
Applying: defined endianness
Applying: fixed warnings in er-coap-13.c
error: patch failed: core/er-coap-13/er-coap-13.c:1231
error: core/er-coap-13/er-coap-13.c: patch does not apply
Patch failed at 0003 fixed warnings in er-coap-13.c
The copy of the patch that failed is found in: .git/rebase-apply/patch
When you have resolved this problem, run "git am --continue".
If you prefer to skip this patch, run "git am --skip" instead.
To restore the original branch and stop patching, run "git am --abort".
/home/florian/Documents/prog/riot_clean2/pkg/pkg.mk:19: recipe for target '/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama/.git-patched' failed
make[1]: *** [/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama/.git-patched] Error 128
/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/../../Makefile.include:352: recipe for target 'pkg-prepare' failed
make: [pkg-prepare] Error 2 (ignored)
"make" -C /home/florian/Documents/prog/riot_clean2/pkg/wakaama
git -C /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama checkout -f 69a32cfae39f66fe4eec4cc8d1cd48ced7ad447c
Warning: you are leaving 2 commits behind, not connected to
any of your branches:
  9416404 defined endianness
  8c72bde changing header location
If you want to keep them by creating a new branch, this may be a good time
to do so with:
 git branch <new-branch-name> 9416404
HEAD is now at 69a32cf... Merge pull request #57 from eclipse/fix_observe
git -C /home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama am --ignore-whitespace "/home/florian/Documents/prog/riot_clean2/pkg/wakaama"/patches/*.patch
fatal: previous rebase directory .git/rebase-apply still exists but mbox given.
/home/florian/Documents/prog/riot_clean2/pkg/pkg.mk:19: recipe for target '/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama/.git-patched' failed
make[1]: *** [/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/pkg/native/wakaama/.git-patched] Error 128
/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/../../Makefile.include:355: recipe for target '/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/native/wakaama.a' failed
make: *** [/home/florian/Documents/prog/riot_clean2/examples/gnrc_wakaama/bin/native/wakaama.a] Error 2

miri64, 2017-11-22:
Who is the maintainer for wakaama at the moment? @kaspar030? @haukepetersen?

florian-jean, 2017-11-22:
I compiled it by changing 
In the makefile of pkg wakaama

> cp $(PKG_BUILDDIR)/core/er-coap-13/*.c $(PKG_TEMP_DIR)/core/er-coap-13/*.h $(PKG_BUILDDIR)

TO

> cp $(PKG_BUILDDIR)/core/er-coap-13/*.c $(PKG_BUILDDIR)/core/er-coap-13/*.h $(PKG_BUILDDIR)

AND in pkg/wakaama/patches/0003-fixed-warnings-in-er-coap-13.c.patch

I removed 

> @@ -1231,7 +1234,7 @@ coap_get_header_size(void *packet, uint32_t *size)
   coap_packet_t *const coap_pkt = (coap_packet_t *) packet;
   if (!IS_OPTION(coap_pkt, COAP_OPTION_SIZE)) return 0;
-
+
   *size = coap_pkt->size;
   return 1;
 }

Because this part generates the error and the original file has the exact same code

haukepetersen, 2017-11-22:
@miri64 I think there is currently no maintainer... I did mentor the project introducing wakaama some time ago, but I don't really feel responsible...

@florian-jean: would you please be so kind a provide a fix for your findings? It seems to me that the package system has undergone some updates over the last years, but `wakaama` is never build (no test application available?) by the CI, so it has not been kept up to date...

florian-jean, 2017-11-22:
@haukepetersen I ll provide a fix.
I m working on making a client out of it. If I succeed I ll make an exemple/test out of it in the same time

Yes, the wakaama cloned by the makefile is A LOT (2 years I guess) behind the current repository. It seems to be quite a bit of a work in order to make all the patches

florian-jean, 2017-11-22:
I was able to make the test as client of wakaama (old version) running in riot, and connect to a test server of wakaama (new version)

Files for the Riot part is attatched. (Compiled with WERROR=0, not clean)

For running the wakaama server
sudo git clone https://github.com/eclipse/wakaama
cd wakaama
sudo cmake tests/server
sudo make
./lwm2mserver

I ll see if it is not too much work to adapt the new version in order to make it work with RIOT.
Also, in order to make the code clean, wakaama define some pre processing variables using cmake. I included them in the makefile of the exemple directly which, I guess is not a good way to do

[riot.zip](https://github.com/RIOT-OS/RIOT/files/1496361/riot.zip)

florian-jean, 2017-11-23:
@haukepetersen  @miri64 
Do you know the tools the patch of wakaama have been generated with ?

miri64, 2017-11-23:
Look like git patches to me:

```bash
man git format-patch
git format-patch <base commit>
```

miri64, 2017-11-23:
(If you want to use the old patches as a basis you need to apply them first of course:

```sh
git am <path-to-riot>/pkg/wakaama/patches/*.patch
```

florian-jean, 2017-11-24:
@miri64 Thanks, I was able to generate the patch for the new version of wakaama (at least for the core and examples)

In order to generates the tests/examples for riot, I need to do conditional inclusion of some folder of wakaama.

So I need the Riot's example Makefile to communicate (just a flag) with the wakaama RIOT's pkg Makefile.

Apparently the CFLAGS are not shared between makefile. Any idea how to make this ?

miri64, 2017-11-24:
>
> So I need the Riot's example Makefile to communicate (just a flag) with
> the wakaama RIOT's pkg Makefile.
>
> Apparently the CFLAGS are not shared between makefile. Any idea how to
> make this ?
>

`export CFLAGS` should do that... But maybe @cladmi can help with that
better.

florian-jean, 2017-11-24:
@miri64 Yep export works. I created a special var in order not modify the behavior of the sub-make.

I'm now struggling with making the conditional statement in the makefile. I have not deal with makefile much up to know (only ultra basic ones). 

I would like to put the conditional statement in the recipe (so interpreted as sh to my understanding) but it doesn't seem to interpret it right. I ll figure it out

cladmi, 2017-11-24:
`CFLAGS` should already be shared between modules because of https://github.com/RIOT-OS/RIOT/blob/2017.10/makefiles/vars.inc.mk#L40

However, all the '-DFOO="bar"' are removed from CFLAGS and put into a common header https://github.com/RIOT-OS/RIOT/blob/2017.10/Makefile.include#L554
So not visible from Makefile but only from the source files, this may be your problem.

@florian-jean what is the "special" case you want to add ? do you have a dirty branch somewhere so I could help ?

florian-jean, 2017-11-24:
@cladmi Well, that's weird, I couldn't see the flags I added to  CFLAGS from the other makefile debug output.

I ended up exporting a var by myself

My branch is local. I ll push it to my git on monday

florian-jean, 2017-11-27:
@cladmi 
So I did a dirty repo 
https://github.com/florian-jean/RIOT

Please see the makefile from tests/wakaama/client
What I did in the pkg/wakaama/Makefile is that I compile others module depending on the var WAKAAMA_EXAMPLE exported by the application makefile.

That enable me to keep the default tree of directories from wakaama and build the examples to incorporate them to riot.
However, I have hard time linking the compiled .c files whatever inclusion I use (USEMODULE, LDFLAGS -L&-l or INCLUDES)

to test go in tests/wakaama/client
sudo rm -rf bin && sudo WERROR=0 make clean all
I still put the WERROR=0 because there is some files left that I need to clean

florian-jean, 2017-11-27:
@cladmi 

I can generate a few library from the makefile
wakaama (the main library from the compilation of the package)
shared_wakaama (the library of tools shared between the exemples)
client_wakaama (the library from the exemple)

I can see them all in the build directory however I can load only wakaama.a through either USEPACKAGE or the INCLUDE += $RIOTPKG/wakaama

Is there any var to access the build dir from the makefile ? Because from it I can use the LDFLAGS and it should work

cladmi, 2017-11-28:
Thanks, I will look at it and get back to you.

cladmi, 2017-11-29:
From a first look, I will refine later:
* first you changed permissions of all files in RIOT to 0755 and this should not be committed :)
* You are currently defining the modules in wakaama/examples/subdirs only when used, and they should always defined. And its easier to just copy a Makefile in them.
    *  the module should have a name prefixed by wakaama `wakaama_example_server` to simplify the next steps
* The configuration to say to include `example_server` should be done through USEMODULE, if you want to include the server module, you need to do `USEMODULE += wakaama_example_server` it will tell the build system to also include `wakaama_example_server.a` when linking.
* Currently you are building the packages in the `patch` step, there is a real solution to this :).
  To build these new modules, in the main Makefile added in the cloned `wakaama` directory, the examples modules build path should be added to the DIRS variable if needed:
    ```
    ifneq (,$(filter wakaama_example_server, $(USEMODULE)))
      DIRS += examples/server
    endif
    ```
    (like in libfixmath Makefile configured through patches: https://github.com/RIOT-OS/RIOT/blob/master/pkg/libfixmath/patches/0001-Move-to-RIOT-Makefiles.patch#L19)

Regarding the INCLUDES variable, I would also do some cleanup maybe but it would require me more analysis.

cladmi, 2017-11-30:
The main issue here is now fixed by merging https://github.com/RIOT-OS/RIOT/pull/8148 so I will close it.
Thank you for raising this problem.

@florian-jean I invite you to open a dedicated PR for adding the wakaama examples.
It will be easier to review there and could synchronize with @moenoel who was also interested to add some stuff to.
You can mention me when you open it, this way I will remember to review it :)

closed 2017-11-30 by cladmi