RIOT-OS/RIOT #8115: cc110x: cc110x_state_to_text() is not defined

hahchenchen, 2017-11-22:
The `cc110x_state_to_text()` is used in cc110x-rxtx.c when enable debug but not defined: https://github.com/RIOT-OS/RIOT/blob/master/drivers/cc110x/cc110x-rxtx.c#L256

So how about remove this function and modify like this:
 `           DEBUG("cc110x: invalid state for sending: %s\n",`
  `                  dev->radio_state==RADIO_RX_BUSY?"RADIO_RX_BUSY":"RADIO_TX_BUSY");`
  `          return -EAGAIN;`
 `   }`

PeterKietzmann, 2017-11-23:
@hahchenchen what you're saying is that `cc110x_state_to_text` is not implemented, right?  If this is the case, this needs to be fixed. Your proposal only involves send and receive busy states. Are there no other possible states?

hahchenchen, 2017-11-23:
Yes, there are no other possible  states in this function:`cc110x_send`

PeterKietzmann, 2017-11-23:
@hahchenchen even no "IDLE" state or something? In that case a solution as proposed seems reasonable. Would you provide a fix?

hahchenchen, 2017-11-26:
No, i would like to provide a fix for it after i fix the spi issue.

maribu, 2018-05-21:
@hahchenchen: Does [this merged PR](https://github.com/RIOT-OS/RIOT/pull/9154) fix your SPI issue?

Kind regards,
Marian

hahchenchen, 2018-05-26:
Yes, it does fix SPI issue.