RIOT-OS/RIOT #8117: Message Queue (cib) breaks on write counter overflow

mlohse, 2017-11-22:
After running for ~4days a thread reproducibly keeps loosing messages sent from an interrupt handler ('msg_send_int') and from itself.
When debugging I noticed, that a message comes through when the target thread was waiting in 'msg_receive' (state 'STATUS_RECEIVE_BLOCKED'). However, when it was to be queued (e.g. because the thread was already active) it gets lost.

Digging deeper I found that the "Circular integer buffer interface" (core/include/cib.h) which is used by the message queue does not handle integer overflows correctly. Write and read counters are incremented but never decremented or reset to 0. So it is just a matter of time until they overflow.
The overflow itself is not a problem if handled correctly, like in 'cib_avail'. However 'cib_get' and 'cib_peek' are using a simple "greater then"-comparison instead of calling 'cib_avail', so the buffer breaks when 'write_count' overflows:

1. Consider the following state for cib_t (buffer is empty, size = 4):
read_count = 0xFFFFFFFF
write_count=0xFFFFFFFF
mask = 0x00000003

2. Call 'cib_put'
-> returns write index: 0x00000003 (now 'write_count' overflows to 0)

3. Call 'cib_get'
-> returns read index: -1 (expected: 0x00000003)
Now the buffer is broken :(

Suggested fix:
Replace 
`if (cib->write_count > cib->read_count) {`
with
`if (cib_avail(cib)) {`
in both 'cib_get' and 'cib_peek'.

kaspar030, 2017-11-22:
nice catch!

kaspar030, 2017-11-22:
@mlohse I guess it takes 4 days to verify that #8119 solves the issue? ;)

mlohse, 2017-11-22:
@kaspar030 Testing for this particular bug on a release-build will take 5 days minimum + another 3 weeks to run a full regression test with different hardware configurations - if everything goes smooth.
I'll keep you posted.... ;)

closed 2017-11-23 by kaspar030

mlohse, 2017-12-05:
Ok, tests against this bug have run through successfully. Regression tests are still in progress, but I doubt that there will be any side-effects... fixed for me :)

PR #8119 (2017-11-23): core: cib: fix overflow handling
```cib_get()``` and ```cib_peek()``` did not handle overflows correctly.

Fixes #8117.