RIOT-OS/RIOT #8118: at86rf2xx: netdev reset

fjmolinas, 2017-11-22:
While using the gnrc_networking example on master I try to send a pkt after resetting the radio by using `ifconfig 7 set state reset`, after resetting UDP packet's don't go out anymore. This doesn't happen when using other radios,  for example mrf24j40. I used samr21-xpro for this.

PeterKietzmann, 2017-11-23:
Intentions to fix it :-)?

kYc0o, 2018-07-19:
@bergzand I think you've fixed this isn't it?

bergzand, 2018-07-19:
Nope, issue still exists

bergzand, 2018-07-19:
The other thing I noticed is that the radio switched to using the short 802154 mac address.

bergzand, 2018-07-19:
> The other thing I noticed is that the radio switched to using the short 802154 mac address.

So this one is a bit more troublesome. On initialization, `nib.c` issues a netopt set call to set the used source MAC address to the long address. In the netdev layer this is just a flag. On a `NETOPT_STATE_RESET` call, this flag is cleared again. the `nib.c` never received the netopt call, and thus the address length is never reapplied.

I think the best fix here is to have the `NETOPT_STATE_RESET` call reset the radio device to the current settings/flags, independent of how the radio is configured when the netopt call is submitted.

miri64, 2018-07-19:
Arghs... Mh... Maybe move the re-initialization to long addresses into a common reset call (#9531) iff `gnrc_sixlowpan` is present?

miri64, 2018-07-31:
From the comments there I'm not sure #9616 actually fixed it, so I'm reopening for now

PeterKietzmann, 2018-07-31:
It does fix the issue described above. However, there are still other issues currently which look similar. I opened a new issue for that #9656. Agree to close this again?

PeterKietzmann, 2018-07-31:
Furthermore, this is not atrf2xx specific (anymore)

miri64, 2018-07-31:
> Agree to close this again?

Ok, closing in favor of #9656.

PR #9616 (2018-07-31): at86rf2xx: Always set channel on device
### Contribution description

This removes the check if the current configured channel equals the new
channel. This check prevents the at86rf2xx channel to be configured
after a reset which causes the radio to be non-functional after a
NETOPT_STATE_RESET.

### Issues/PRs references

fixes #8118

### Testing

Branch is intentionally not based on the latest master to prevent #9577/#9606 from causing issues.

The issue that causes gnrc_netif to switch to short source MAC addresses is not resolved here.
closed 2018-07-31 by miri64