RIOT-OS/RIOT #8121: rpl: ENABLE_DEBUG=1 doesn't compile

bergzand, 2017-11-22:
Not caused by #7925, but doesn't make debugging easier.

## How to reproduce:

set ENABLE_DEBUG to (1) in `sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c`

## Expected result:

Code compiles

## Result

Multiple errors, some about unused vars, some about macro redefined.

(Mostly a reminder to myself to clean this up tomorrow or so.

miri64, 2017-11-23:
See #5166 as a potential permanent fix ;-)

miri64, 2017-11-23:
RPL DEBUG seems to be pretty broken in general... (ENABLE_DEBUG defines in header files, what?)

bergzand, 2017-11-23:
> (ENABLE_DEBUG defines in header files, what?)

My thoughts exactly. :)

PeterKietzmann, 2019-07-23:
@bergzand is this still the case almost 2 years later and with #5166 being merged?

bergzand, 2019-07-23:
@PeterKietzmann Unfortunately, I can still reproduce the issue on latest master when enabling `DEBUG` in `sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c`:

```
Building application "gnrc_networking" for "native" with MCU "native".

"make" -C /home/hydrazine/dev/RIOT/boards/native
"make" -C /home/hydrazine/dev/RIOT/core
"make" -C /home/hydrazine/dev/RIOT/cpu/native
"make" -C /home/hydrazine/dev/RIOT/drivers
"make" -C /home/hydrazine/dev/RIOT/sys
"make" -C /home/hydrazine/dev/RIOT/boards/native/drivers
"make" -C /home/hydrazine/dev/RIOT/cpu/native/netdev_tap
"make" -C /home/hydrazine/dev/RIOT/cpu/native/periph
"make" -C /home/hydrazine/dev/RIOT/cpu/native/vfs
"make" -C /home/hydrazine/dev/RIOT/drivers/netdev_eth
"make" -C /home/hydrazine/dev/RIOT/drivers/periph_common
"make" -C /home/hydrazine/dev/RIOT/sys/auto_init
"make" -C /home/hydrazine/dev/RIOT/sys/div
"make" -C /home/hydrazine/dev/RIOT/sys/evtimer
"make" -C /home/hydrazine/dev/RIOT/sys/fmt
"make" -C /home/hydrazine/dev/RIOT/sys/iolist
"make" -C /home/hydrazine/dev/RIOT/sys/luid
"make" -C /home/hydrazine/dev/RIOT/sys/net/crosslayer/inet_csum
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc
"make" -C /home/hydrazine/dev/RIOT/sys/net/link_layer/l2util
"make" -C /home/hydrazine/dev/RIOT/sys/net/network_layer/icmpv6
"make" -C /home/hydrazine/dev/RIOT/sys/net/network_layer/ipv6/addr
"make" -C /home/hydrazine/dev/RIOT/sys/net/network_layer/ipv6/hdr
"make" -C /home/hydrazine/dev/RIOT/sys/net/transport_layer/udp
"make" -C /home/hydrazine/dev/RIOT/sys/od
"make" -C /home/hydrazine/dev/RIOT/sys/auto_init/netif
"make" -C /home/hydrazine/dev/RIOT/sys/ps
"make" -C /home/hydrazine/dev/RIOT/sys/random
"make" -C /home/hydrazine/dev/RIOT/sys/shell
"make" -C /home/hydrazine/dev/RIOT/sys/shell/commands
"make" -C /home/hydrazine/dev/RIOT/sys/trickle
"make" -C /home/hydrazine/dev/RIOT/sys/xtimer
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/netapi
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/netif
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/netreg
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/icmpv6
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/icmpv6/echo
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/icmpv6/error
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/hdr
"make" -C /home/hydrazine/dev/RIOT/sys/random/tinymt32
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ipv6/nib
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/network_layer/ndp
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/pkt
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/pktbuf
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/pktbuf_static
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/pktdump
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/netif/ethernet
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/routing/rpl
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/netif/hdr
"make" -C /home/hydrazine/dev/RIOT/sys/net/gnrc/transport_layer/udp
/home/hydrazine/dev/RIOT/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c:46: error: "ENABLE_DEBUG" redefined [-Werror]
 #define ENABLE_DEBUG    (1)
 
In file included from /home/hydrazine/dev/RIOT/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c:37:
/home/hydrazine/dev/RIOT/sys/net/gnrc/routing/rpl/gnrc_rpl_internal/validation.h:29: note: this is the location of the previous definition
 #define ENABLE_DEBUG    (0)
 
cc1: all warnings being treated as errors
make[4]: *** [/home/hydrazine/dev/RIOT/Makefile.base:83: /home/hydrazine/dev/RIOT/examples/gnrc_networking/bin/native/gnrc_rpl/gnrc_rpl_control_messages.o] Error 1
make[3]: *** [/home/hydrazine/dev/RIOT/Makefile.base:20: ALL--/home/hydrazine/dev/RIOT/sys/net/gnrc/routing/rpl] Error 2
make[2]: *** [/home/hydrazine/dev/RIOT/Makefile.base:20: ALL--/home/hydrazine/dev/RIOT/sys/net/gnrc] Error 2
make[1]: *** [/home/hydrazine/dev/RIOT/Makefile.base:20: ALL--/home/hydrazine/dev/RIOT/sys] Error 2
make: *** [/home/hydrazine/dev/RIOT/examples/gnrc_networking/../../Makefile.include:485: /home/hydrazine/dev/RIOT/examples/gnrc_networking/bin/native/application_gnrc_networking.a] Error 2
make: Leaving directory '/home/hydrazine/dev/RIOT/examples/gnrc_networking'
```

bergzand, 2019-08-16:
Fixed now that #9852 is merged

closed 2019-08-16 by bergzand