RIOT-OS/RIOT #8122: doxygen: riot.css modified by 'make doc'

kaspar030, 2017-11-23:
Calling ```make doc``` modifies riot.css in-place, making it show up in git's changed files list.
That's a little annoying. Can that be avoided?

aabadie, 2017-11-23:
> Can that be avoided?

I had the same problem and I fixed it by uninstalling lessc from npm and install python-lesscpy via apt-get instead.

miri64, 2017-11-23:
> Can that be avoided?

An alternative would be to switch over from Bootstrap 3 to Bootstrap 4, which uses SASS instead of LESS (which does the compilation live in-browser). Would need some time however.

jcarrano, 2019-06-27:
See #10249 for a fix to this issue.

aabadie, 2020-07-16:
Is this still an issue ?

miri64, 2020-07-16:
We did not change how the `riot.css` is generated, so I'd say, yes. Unless people don't see this as an issue anymore :-D.

aabadie, 2020-07-16:
I also had this problem in the past but now it doesn't show up anymore. That's why I'm asking.

miri64, 2020-07-16:
Do you still have LESS compiler installed? Does it incidentally generate the same CSS as the one at hand?

aabadie, 2020-07-16:
> Do you still have LESS compiler installed?

Ah, that might be that: I don't have LESS compiler :)

closed 2023-05-25 by bors[bot]