RIOT-OS/RIOT #8125: spi code not linked on cc2538 target

geith, 2017-11-23:
The latest master version (b2b2071142c77d39f9fd26535fbfcff2b1197b9f) does not link the spi code (and some other parts too) for a cc2538 target (e.g. _remote_revb_).

When building a minimal test program with (see below) with the latest version for the _remote_revb_ target, I get the following error message while it compiles for older RIOT versions.
```
$ make BOARD=remote-revb
...
/mnt/home_ext/geith/src/riot-os/spi-test/bin/remote-revb/default.a(main.o): In function `main':
/mnt/home_ext/geith/src/riot-os/spi-test/main.c:6: undefined reference to `spi_init'
collect2: error: ld returned 1 exit status
/mnt/home_ext/geith/src/riot-os/spi-test/../RIOT/Makefile.include:306: recipe for target 'link' failed
```
```c
#include "periph/spi.h"

int main(void)
{
    spi_t spi = SPI_DEV(0);
    spi_init(spi);

    return 0;
}
```

The SPI related code doesn't show up in the build directory (latest RIOT version):
```
$ ls bin/remote-revb/periph/
adc.d  adc.o  gpio.d  gpio.o  pm.d  pm.o  uart.d  uart.o
```
On older RIOT versions (e.g. 2017.10), the build directory looks like this:
```
$ ls bin/remote-revb/periph/
adc.d  gpio.d  hwrng.d  i2c.d  pm.d  spi.d  timer.d  uart.d
adc.o  gpio.o  hwrng.o  i2c.o  pm.o  spi.o  timer.o  uart.o
```

dylad, 2017-11-23:
Hi @geith 
I tried to compile test/periph_spi on the lastest master (6b661be) and everything is fine :  

```
 make BOARD=remote-revb
Building application "tests_periph_spi" for "remote-revb" with MCU "cc2538".

"make" -C /home/dylan/Software/RIOT/boards/remote-revb
"make" -C /home/dylan/Software/RIOT/boards/remote-common
"make" -C /home/dylan/Software/RIOT/core
"make" -C /home/dylan/Software/RIOT/cpu/cc2538
"make" -C /home/dylan/Software/RIOT/cpu/cc2538/periph
"make" -C /home/dylan/Software/RIOT/cpu/cortexm_common
"make" -C /home/dylan/Software/RIOT/cpu/cortexm_common/periph
"make" -C /home/dylan/Software/RIOT/drivers
"make" -C /home/dylan/Software/RIOT/drivers/periph_common
"make" -C /home/dylan/Software/RIOT/sys
"make" -C /home/dylan/Software/RIOT/sys/auto_init
"make" -C /home/dylan/Software/RIOT/sys/div
"make" -C /home/dylan/Software/RIOT/sys/isrpipe
"make" -C /home/dylan/Software/RIOT/sys/newlib_syscalls_default
"make" -C /home/dylan/Software/RIOT/sys/shell
"make" -C /home/dylan/Software/RIOT/sys/shell/commands
"make" -C /home/dylan/Software/RIOT/sys/tsrb
"make" -C /home/dylan/Software/RIOT/sys/uart_stdio
"make" -C /home/dylan/Software/RIOT/sys/xtimer
   text	   data	    bss	    dec	    hex	filename
  16292	    500	   3648	  20440	   4fd8	/home/dylan/Software/RIOT/tests/periph_spi/bin/remote-revb/tests_periph_spi.elf
```
```
ls bin/remote-revb/periph/
gpio.d  gpio.o  pm.d  pm.o  spi.d  spi.o  timer.d  timer.o  uart.d  uart.o
```

Could you provide your GCC version and try 
`make clean all BOARD=remote-revb`

geith, 2017-11-23:
Hi @dylad,

no, still the same... (I used `rm -rf bin/` before).

I'm using Ubuntu 17.10, gcc (arm-none-eabi-gcc) has the version 5.4.1.

dylad, 2017-11-23:
I'm using `gcc version 5.4.1 20160919 (release) `

`/mnt/home_ext/geith/src/riot-os/spi-test` 
You're not using the default RIOT, isn't it ?

dylad, 2017-11-23:
Also could you ensure you have `FEATURES_REQUIRED = periph_spi` within your Makefile application
and `FEATURES_PROVIDED += periph_spi` in your Makefile.features (from your board directory)

geith, 2017-11-23:
Hi @dylad,
For this test, I'm using the default RIOT (https://github.com/RIOT-OS/RIOT.git) ant switching between the current master branch and 2017.10.
My test program is located in a separate directory besides the RIOT folder (`RIOTBASE ?= $(CURDIR)/../RIOT`).

I just tested that on another machine with Debian 9 (gcc has the version 5.4.1 too) with the same problem.

The `FEATURES_REQUIRED = periph_spi` statement in my makefile helped, now it compiles with the latest master version too.

Thanks a lot!

closed 2017-11-23 by geith

dylad, 2017-11-23:
> now it compiles with the latest master version too

Glad to hear !