RIOT-OS/RIOT #8129: gnrc_rpl: Don't add routes without a next hop to DAO

bergzand, 2017-11-23:
Minor quirk int the new gnrc_rpl code, causes routing tables such as:

```
2017-11-23 17:54:39,861 - INFO #  nib route
2017-11-23 17:54:39,867 - INFO # 2001:db8::/64 dev #7
2017-11-23 17:54:39,873 - INFO # 2001:db8::1217:7220:1043:1a4e/128 via fe80::1217:7220:1043:1a4e dev #7
2017-11-23 17:54:39,878 - INFO # 2001:db8::/64 via fe80::1217:7220:1043:1a4e dev #7
2017-11-23 17:54:39,879 - INFO # default* via fe80::1116:7226:122d:1402 dev #7
```

Note entry number 1 and 3. Fix in a few minutes or hours.

miri64, 2017-11-23:
Err, entry one is valid and the prefix list entry. It is however a bug, that the NIB doesn't update PL entry to a route entry instead of creating a whole new entry. I can look into that as well.

bergzand, 2017-11-23:
> Err, entry one is valid and the prefix list entry

Yes, blame that on some my bad wording of the issue. What I tried to say is that entry number 3 is invalid (as far as I know), and overlaps/conflicts with entry number 1 which is correct.

miri64, 2017-11-23:
> Yes, blame that on some my bad wording of the issue. What I tried to say is that entry number 3 is invalid (as far as I know), and overlaps/conflicts with entry number 1 which is correct.

Yes and no. Entry 1 is supposed to be updated to be equal to entry 3.

miri64, 2017-11-23:
Working on some unittests to support my hunch ;-).

miri64, 2017-11-23:
Mhhhhh.... I did that, but it didn't fail...

```diff
diff --git a/tests/unittests/tests-gnrc_ipv6_nib/tests-gnrc_ipv6_nib-ft.c b/tests/unittests/tests-gnrc_ipv6_nib/tests-gnrc_ipv6_nib-ft.c
index 5dfa4fa..bffdd7b 100644
--- a/tests/unittests/tests-gnrc_ipv6_nib/tests-gnrc_ipv6_nib-ft.c
+++ b/tests/unittests/tests-gnrc_ipv6_nib/tests-gnrc_ipv6_nib-ft.c
@@ -545,6 +545,35 @@ static void test_nib_ft_add__success_overwrite_unspecified(void)
 }
 
 /*
+ * Creates a prefix list entry and then adds another with equal prefix and
+ * interface t the last, but with a next hop address.
+ * and interface to the last, but with a next hop address
+ * Expected result: there should only be one route (with the configuration of
+ * the later added route), prefix list should also contain one entry
+ */
+static void test_nib_ft_add__success_merge_pl(void)
+{
+    gnrc_ipv6_nib_ft_t fte;
+    gnrc_ipv6_nib_pl_t ple;
+    void *iter_state = NULL;
+    static const ipv6_addr_t dst = { .u64 = { { .u8 = GLOBAL_PREFIX } } };
+    static const ipv6_addr_t next_hop = { .u64 = { { .u8 = LINK_LOCAL_PREFIX },
+                                                 { .u64 = TEST_UINT64 } } };
+
+    TEST_ASSERT_EQUAL_INT(0, gnrc_ipv6_nib_pl_set(IFACE, &dst, GLOBAL_PREFIX_LEN,
+                                                  UINT32_MAX, UINT32_MAX));
+    TEST_ASSERT_EQUAL_INT(0, gnrc_ipv6_nib_ft_add(&dst, GLOBAL_PREFIX_LEN,
+                                                  &next_hop, IFACE, 0));
+    TEST_ASSERT(gnrc_ipv6_nib_ft_iter(NULL, 0, &iter_state, &fte));
+    TEST_ASSERT(ipv6_addr_equal(&fte.next_hop, &next_hop));
+    TEST_ASSERT(!gnrc_ipv6_nib_ft_iter(NULL, 0, &iter_state, &fte));
+    iter_state = NULL;
+    TEST_ASSERT(gnrc_ipv6_nib_pl_iter(0, &iter_state, &ple));
+    TEST_ASSERT(ipv6_addr_equal(&ple.pfx, &dst));
+    TEST_ASSERT_EQUAL_INT(GLOBAL_PREFIX_LEN, ple.pfx_len);
+}
+
+/*
  * Creates a route
  * Expected result: a new entry should exist and contain the given prefix,
  * interface, and lifetimes
@@ -744,6 +773,7 @@ Test *tests_gnrc_ipv6_nib_ft_tests(void)
         new_TestFixture(test_nib_ft_add__ENOMEM_diff_dst_dst_len_next_hop_iface),
         new_TestFixture(test_nib_ft_add__success_duplicate),
         new_TestFixture(test_nib_ft_add__success_overwrite_unspecified),
+        new_TestFixture(test_nib_ft_add__success_merge_pl),
         new_TestFixture(test_nib_ft_add__success),
         new_TestFixture(test_nib_ft_add__success_dr),
         new_TestFixture(test_nib_ft_del__unknown),
```

Won't add it as such.

miri64, 2017-11-23:
(it's pretty similar to `test_nib_ft_add__success_overwrite_unspecified` anyway)

bergzand, 2017-11-23:
The old RPL code doesn't add the prefix to the DAO packets. I think the fix here is to extend the check on the `fte` [here]( https://github.com/RIOT-OS/RIOT/blob/gnrc_netif2_integration/master/sys/net/gnrc/routing/rpl/gnrc_rpl_control_messages.c#L754) with a check if `!ipv6_addr_is_unspecified(&fte.next_hop)`

bergzand, 2017-11-23:
Maybe we're talking about two different issues here?

miri64, 2017-11-23:
> Maybe we're talking about two different issues here?

Maybe. And yes, your fix sounds reasonable.

miri64, 2017-11-23:
Fixed in #8132.

PR #8132 (2017-11-23): gnrc_rpl: Exclude routes without next hop from DAO packets
The DAO packets contain RPL target options which don't have a next hop on the sender. This causes routes for the prefix to be propagated up the tree. This PR prevents addition of routes with an unspecified next hop to be added to the target options.

Fixes #8129
closed 2017-11-23 by miri64